cms-gem-daq-project/ctp7_modules #8: Feature Request: Library Generation for HW Version

bdorney, 2017-10-11:
## Brief summary of issue

Based on the discussion we had yesterday for a given CTP7 I understand the following:

1. The host PC issues an RPC call using `xhal`.  This call goes through a function named `myRPCMethod(arg1, arg2, ...)` defined in the `xhal` repo. 
2. The function `myRPCMethod(arg1, arg2, ...)` calls an RPC module method stored in this (`ctp7_modules`) repo called `myRPCMethodModule(const RPCMsg *request, RPCMsg *response)`.
3. The function `myRPCMethodModule(const RPCMsg *request, RPCMsg *response)` calls `myRPCMethodModuleLocal(input1, input2, ...)`.  Where `myRPCMethodModuleLocal(...)` is executed on the CTP7/CTP7-emulator.

The function `myRPCMethod(...)` called from the host PC sets the words that `myRPCMethodModule(...)` needs, which are the same for v3 and v2b.  This may imply we will need to have some dummy and/or translating arguments; e.g. "MSPL" <-> "PULSE_STRETCH".

Based on the discussion yesterday we will have two versions of `myRPCMethodModuleLocal(input1, input2, ...)` each taking the same input parameters but one will be for v2b and one will be for v3.  

From our discussion yesterday a CTP7/CTP7 emulator controlling v2b (v3) electronics will have the library where all the `myRPCMethodModuleLocal(input1, input2, ...)` will be compiled for v2b (v3) electronics behavior, e.g. `libCTP7Modules_v2b.so` or `libCTP7Modules_v3.so`.

What I propose then is the following convention for this repository (`ctp7_modules`):

- Header files are named in a HW independent way, e.g. `include/vfat.h`, `include/optohybrid.h`,
- The *non-local* functions are defined and implemented in the corresponding header file while *local* functions are defined in the header file but implemented in the corresponding HW dependently named src file.
- Source files are named in a HW *dependent way* using the regular expressions `_v2b` and `_v3` just before the file extension but include the HW independent header file, e.g. `src/vfat_v2b.cc` and `src/vfat_v3.cc` which both includes `include/vfat.h`; or `src/optohybrid_v2b.cc` and `src/optohybrid_v3.cc` which both include `include/optohybrid.h`.
- The v2b functionality of any given `myRPCMethodModuleLocal(input1, input2, ...)` is implemented in the corresponding `src/*_v2b.cc` file and the v3 functionality of any given `myRPCMethodModuleLocal(input1, input2, ...)` is implemented in the corresponding `src/*_v3.cc` file.  In each case the order and names of the arguments is identical.
- The Makefile is changed to support additional make options:`make all` makes v2 and v3 libraries, e.g. `libCTP7Modules_v2b.so` or `libCTP7Modules_v3.so`; `make v2b` makes just the v2b library e.g. `libCTP7Modules_v2b.so`; and `make v3` makes just the v3 library e.g. `libCTP7Modules_v3.so`.

Also, I would propose also that rpc header and src files defined in the `xhal` repo are all named in HW independent manner.

I think the following example is illustrative, consider the case for:

- `myRPCMethod(arg1, arg2, ...)` = `DLLEXPORT uint32_t broadcastWrite(uint32_t ohN, char * regName, uint32_t value, uint32_t vfatMask)`
- `myRPCMethodModule(const RPCMsg *request, RPCMsg *response)` = `void broadcastWrite(const RPCMsg *request, RPCMsg *response)`
- `myRPCMethodModuleLocal(input1, input2, ...)` = `void broadcastWriteLocal(lmdb::txn & rtxn, lmdb::dbi & dbi, std::string oh_number, std::string reg_to_write, uint32_t val_to_write, RPCMsg * response, uint32_t mask = 0xFF000000)`

The function `DLLEXPORT uint32_t broadcastWrite(uint32_t ohN, char * regName, uint32_t value, uint32_t vfatMask)` is defined in `xhal/include/xhal/rpc/optohybrid.h`, and implemented in `xhal/src/common/rpc_manager/optohybrid.cc`.

The function `void broadcastWrite(const RPCMsg *request, RPCMsg *response)` is defined and implemented in `ctp7_modules/include/optohybrid.h`.

The function `void broadcastWriteLocal(lmdb::txn & rtxn, lmdb::dbi & dbi, std::string oh_number, std::string reg_to_write, uint32_t val_to_write, RPCMsg * response, uint32_t mask = 0xFF000000)` is defined in `ctp7_modules/include/optohybrid.h` but implemented in two different locations:
- `ctp7_modules/src/optohybrid_v2b.cc`
- `ctp7_modules/src/optohybrid_v3.cc`

The implementation of each of the above is shown specifically below.  Then lines 27-37 of the `ctp7_modules` [Makefile](https://github.com/cms-gem-daq-project/ctp7_modules/blob/master/Makefile) are changed too:

```
SRCS_V2B= $(shell echo ${CTP7_MOD_ROOT}/src/*_v2b.cpp)
SRCS_V3= $(shell echo ${CTP7_MOD_ROOT}/src/*_v3.cpp)
TARGET_LIBS_V2B=$(addprefix lib/,$(patsubst %_v2b.cpp, %_v2b.so, $(notdir $(wildcard ./src/*_v2b.cpp))))
TARGET_LIBS_V3=$(addprefix lib/,$(patsubst %_v3.cpp, %_v3.so, $(notdir $(wildcard ./src/*_v3.cpp))))

all:

lib/optical.so: src/optical.cpp 
	$(CXX) $(CFLAGS) $(INC) $(LDFLAGS) -fPIC -shared -o $@ $< -lwisci2c

$(TARGET_LIBS_V2B): $(SRCS_V2B)
lib/%.so:src/%.cpp
	$(CXX) $(CFLAGS) -std=c++1y -O3 -pthread $(INC) $(LDFLAGS) -fPIC -shared -o $@ $^ -lwisci2c -lxhal_ctp7 -llmdb

$(TARGET_LIBS_V3): $(SRCS_V3)
lib/%.so:src/%.cpp
	$(CXX) $(CFLAGS) -std=c++1y -O3 -pthread $(INC) $(LDFLAGS) -fPIC -shared -o $@ $^ -lwisci2c -lxhal_ctp7 -llmdb

v2b:

lib/optical.so: src/optical.cpp 
	$(CXX) $(CFLAGS) $(INC) $(LDFLAGS) -fPIC -shared -o $@ $< -lwisci2c

$(TARGET_LIBS_V2B): $(SRCS_V2B)
lib/%.so:src/%.cpp
	$(CXX) $(CFLAGS) -std=c++1y -O3 -pthread $(INC) $(LDFLAGS) -fPIC -shared -o $@ $^ -lwisci2c -lxhal_ctp7 -llmdb

v3:

[... 129 lines cut ...]
   for(uint32_t vfatN = 0; vfatN < 24; vfatN++){
     if((notmask >> vfatN) & 0x1){
          reg_name = reg_basename + std::to_string(vfatN)+".CFG_" + reg_to_write;
          writeReg(la->rtxn, la->dbi, reg_name, val_to_write, la->response);
     }
  }
}

extern "C" {
    const char *module_version_key = "optohybrid v1.0.1";
    int module_activity_color = 4;
    void module_init(ModuleManager *modmgr) {
        if (memsvc_open(&memsvc) != 0) {
            LOGGER->log_message(LogManager::ERROR, stdsprintf("Unable to connect to memory service: %s", memsvc_get_last_error(memsvc)));
            LOGGER->log_message(LogManager::ERROR, "Unable to load module");
            return; // Do not register our functions, we depend on memsvc.
        }
        modmgr->register_method("optohybrid", "broadcastWrite", broadcastWrite);
    }
} 
```

## Context (for feature requests)

Need some conventions defined for future development work.

bdorney, 2017-10-11:
After some thought and attempts today to start implementation I the above should be edited. I'll update and make new suggestions tomorrow morning.

jsturdy, 2017-10-11:
To (hopefully) briefly summarize (@mexanick can refine the RPC side, and we could consider putting together a working document for this)
* Current setup
  * `IPBus`
    * `uhal` library providing register access (read/write) to `IPBus` HW endpoints
    * `GEMHwDevice class: wrapper around `uhal` library providing some utility for the way the GEM address table is structured and for creating device connections
    * `Hw<board>` class: extends `GEMHwDevice` to provide functions for specific hardware device (or something equivalent)
    * Independent `python` package using the `uhal` `python` interface directly, but providing many of the same functions through modules
      * Specific tools built on top of this package to do certain tasks (`amc_info_uhal.py`, `ultraScurve.py`, `run_scans.py`, etc.)
  * The RPC based pattern is an amalgamation of this
    * `memsvc` library provides "equivalent" library to `uhal`
    * `rpcsvc` connects this to RPCs
    * `rpc-manager` adds modules in the vein of the `Hw<board>` class but as standalones?
    * In this case there is no underlying `GEMHwDevice`
    * Some porting of the "specific" tools from above was done in the past, but in an effort to not duplicate work, was not used until such a time as it was appropriate (v3 electronics/post stable CMS integration)

From our discussion it seems that the end system should be:
* A library which provides read/write access on registers (`uhal`/`memsvc`)
*  Library/class which provides functions to do operations using these
  * This library will use an RPC message to perform the transaction (whether remote or local)
  * `Hw<board>` be the interface between online software and this lower level world
    * It will either be gutted, because it doesn't fit into the RPC model that Misha has set up, and simply provide forwarding calls to the RPC manager or
    * The the functional definitions, including the RPC portion, will be implemented inside it, but this is not in the way that Misha has currently been using the RPC methodology
  * Exported `python` library with additions as needed (`amc_info.py`, `ultraScurve.py`, `run_scans.py`, etc.)

From long ago discussions initially, the `xhal` library was envisioned as being the `rpc` equivalent of `uhal`, but due to differences in the architecture as a whole, this ended up taking on a different form, and then giving birth to `ctp7_modules`.  Cleaning up and making the whole architecture coherent was, of course, the primary goal of the discussion on Tuesday, and is the goal of this issue I believe

bdorney, 2017-10-12:
From our discussion it seems that the end system should be:

> From our discussion it seems that the end system should be:
> * A library which provides read/write access on registers (`uhal`/`memsvc`)

From my understanding this library is generated in the `xhal` repo and should be `xhal/lib/x86_64/librpcman.so`.  In `xhal` we would have electronics version independent modules that rpc-manager adds (so yeah in the vein of the `Hw<board>` class but as standalones).  These modules exchange rpc request/responses to the CTP7/CTP7-emulator.  

So in my example above I have in `xhal`

- [include/xhal/rpc/optohybrid.h](https://github.com/bdorney/xhal/blob/feature_rpcOHFunctionality/include/xhal/rpc/optohybrid.h)
- [src/common/rpc_manager/optohybrid.cc](https://github.com/bdorney/xhal/blob/feature_rpcOHFunctionality/src/common/rpc_manager/optohybrid.cc)

These require the functions defined in `ctp7_modules`:

- [include/optohybrid.h](https://github.com/bdorney/ctp7_modules/blob/feature_v3OHFeatures/include/optohybrid.h)
- [src/optohybrid.cpp](https://github.com/bdorney/ctp7_modules/blob/feature_v3OHFeatures/src/optohybrid.cpp)

And yeah after further thought we should just have one electronics independent version of the file where each function has the behavior of v2b and v3 electronics coded in and operates on either 1) determining the AMC FW version at runtime, or 2) receives a isV3El flag.

> *  Library/class which provides functions to do operations using these

I am trying to provide this in `cmsgemos/python/utils/registers_xhal.py` as a class `rpcService`.  My idea was that I would have a corresponding set of `Hw<board>` classes in python that would have an instance of `rpcService` as a data member.  I have added to cmsgemos:

- [gempython/utils/registers_xhal.py](https://github.com/bdorney/cmsgemos/blob/feature_RPCInterface/gempython/utils/registers_xhal.py)

> * This library will use an RPC message to perform the transaction (whether remote or local)

For example the class `rpcService` has data members:

- self.broadcastRead(...)
- self.broadcastWrite(...)

Which come from the library mentioned above.

>  * `Hw<board>` be the interface between online software and this lower level world

I am trying to implement this in `cmsgemos` as a python class [HwOptohybrid](https://github.com/bdorney/cmsgemos/blob/6bcc89a2d55499f41aee9047eb80cd387363abd4/gempython/tools/optohybrid_user_functions_xhal.py#L15).

> * It will either be gutted, because it doesn't fit into the RPC model that Misha has set up, and simply provide forwarding calls to the RPC manager or
>    * The the functional definitions, including [...]

I know the long-term goal is that this should all be done in `C++` and then exported using `boost.python` ([Link?](http://www.boost.org/doc/libs/1_65_1/libs/python/doc/html/tutorial/index.html)) but at the time it seemed easier trying to set this up in `gempython` since I am more familiar with that.  I saw that there are just a few functions that need to be truly re-written which, once done, will give an immediately expanded functionality, e.g. `confChamber.py` might work out of the box if an `xhal` implementation of `broadcastRead(...)` and `broadcastWrite(...)` exist since: `writeAllVFATs(...)`, `readAllVFATs(...)`, and `biasAllVFATs(...)` are appear to be just wrappers around the broadcast functions.

My thought in each case was to:

1. Start with the v2b case to confirm the RPC calls worked (e.g. confirm existing behavior), and then
2. Encode the case statement `if (v3el) { ... }` into the corresponding `ctp7_modules` for the v3 functionality.

After a further look at this in my mind then there would not be `vfat3.cc` and `vfat2.cc` file but just a `vfat.cc` file.

Hopefully I have understood all this?

mexanick, 2017-10-12:
@bdorney , we had a bit of extended discussion with @jsturdy after our big meeting. While what you have written above is correct in case of standalone implementation, we certainly should look towards integration with the cmsgemos. 
I think we should split the development process into two phases:
1. right now, use the rpc_manager library standalone as is. We are not going to develop much there, mainly bugfixes, since the required functionality to qualify the v3 electronics is there (scans). 
2. we will start development of RPC-related things under cmsgemos. The GEMHwDevice class will become an abstract interface using overloaded methods for V2, V3 and GLIB/CTP7 connections. It will be extended by specific device classes as @jsturdy wrote above. 

We definitely should start a tech-spec document with all this stuff. I'll try to setup a sharelatex starting page.

jsturdy, 2020-01-15:
Superseded by templated framework, yes @mexanick?

mexanick, 2020-01-16:
Certainly!

closed 2020-01-16 by mexanick