cms-gem-daq-project/ctp7_modules #14: Feature Request: RPC Method for Setting Channel Registers

bdorney, 2018-01-08:
## Brief summary of issue

Right now channel registers for a given VFAT are set by looping over all channels in the VFAT and then issuing an RPC call for *each* channel.  See:

- [confChamber.py](https://github.com/bdorney/vfatqc-python-scripts/blob/85da7e9f9710067c4f259b37ffb85d01bc96438d/confChamber.py#L117)
- [setChannelRegister(...)](https://github.com/bdorney/cmsgemos/blob/feature_RPCInterface/gempython/tools/vfat_user_functions_xhal.py#L121)

As you might imagine, this is a super slow process when configuring.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

There should be a single RPC call per link that is executed and sets all the channel registers.  For each channel register there should be an entry in the input RPC message of type `word_array` that is 3072 entries in length.

So for VFAT3 the following word arrays would exist (all of type `uint32_t`):
- `calPulseEnable[3072]`
- `mask[3072]`
- `trimARM[3072]`
- `trimZCC[3072]`

The user then just supplies these arrays as input to the xhal method that sets these as `word_array` in the RPC message.

The position in the array would be uniquely linked to a `(vfat_pos, ch)` pair via the equation:

```
idx = vfat_pos * 128 + ch
```

Where `vfat_pos` is 0 to 23 and `ch` is 0 to 127.

Additionally RPC method could also set in the RPC Response a set of `word_arrays` that are the readback values.  One could then compare the values to be written and the current readback values outside the RPC method.

## Current Behavior

Right now 3072 RPC calls are performed when setting VFAT3 channel registers.

## Context (for feature requests)

Configuring channel registers is super slow.

## Your Environment

* Version used:
* Shell used:

bdorney, 2018-01-08:
@mexanick and @jsturdy what do you think of the proposed implementation?

mexanick, 2018-01-09:
I totally agree that the current situation is very inefficient. However I have slightly different proposal of implementation. Our operational experience says that we have high chances of having the chips which are masked completely (or don't work for a number of reasons), especially at the test/dev stage. Given this I don't like an idea to have the arrays of size 3072 containing all the values per GEB. I would rather put N arrays of size 128, where N is the number of chips we want to program. Actually, given the current implementation of the RPCMsg, it shouldn't be an array, but a vector of uint32_t. Furthermore, the word_array struct of the RPCMsg has a name, so we can put there the chip name in a certain format. 
Example:
_RPC request:_
```
RPCMsg *req;
std::vector<uint32_t> masks_per_chip;
std::string arr_name = "masks";
for (auto & chip: chips) {
    masks_per_chip = chip->getMasks();
    req->set_word_array(arr_name+std::to_string(chip->getPosition()),masks_per_chip);
}
```
_RPC call implentation at the CTP7:_
```
std::string key;
std::vector<uint32_t> vals
for (int i=0; i<24; i++) {
    //set masks
    key = "masks"
    key += std::to_string(i);
    if (req->get_key_exist(key)) {
        vals = req->get_word_array(key);
        writeVFATChanReg("mask", i, vals);
        // possibly read back and form the rsp here
    }
}
```

bdorney, 2018-01-09:
Yeah I like this idea better, then we won’t loop over masked chips. Good point. Will try to work on it this week.

mexanick, 2018-01-09:
Before you start on actual implementation, I'd like to see/discuss with @jsturdy the interfaces. 
I want to see the complete use case diagram for channel registers setting, including whether we **always** want to set all of them (calPulse, mask, trimARM and trimZCC) or not. It might be better/simpler/more straightforward to implement one RPC call per chan reg type. In such case we won't need to hardcode the chan reg type in the word array key.

bdorney, 2018-01-09:
The original idea of the VFAT3 design team is that a given channel uses the AND of the coincidence of the arming (ARM) and zero crossing (ZCC) comparators.  So in actual operation we would need to write both the trimARM and trimZCC values.  Right now however we are only using the arming comparator.

I could see the following use cases:
1. Writing calpulse to 0 for all channels (this is already implemented: [stopCalPulses2AllChannels(...)](https://github.com/cms-gem-daq-project/ctp7_modules/blob/3d5e6c60f7aafb7dcf5dc2a790fe93f008dd705e/include/optohybrid.h#L500) both local and non-local forms)
2. Writing calpulse to 1 for vfat N channel ch
3. Writing calpulse to 1 for all VFATs channel ch (but this is just lopping the above case)
4. Writing mask, trimARM, and trimZCC settings for configuration.

So at most this would need three distinct functions, one of which is already implemented (assuming we are satisfied with the implementation).