cms-gem-daq-project/ctp7_modules #19: Implement status/monitoring methods

mexanick, 2018-02-13:
We need a set of methods working with V3 electronics to report the CTP7-OH communication status, DAQ state, TTC state etc. 
## Brief summary of issue

In order to implemented the functionality described above an input from FW developers is required considering the set of registers and FW modules which has to be monitored.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

Expected behavior is similar to existing V2-based methods
## Current Behavior

None

## Context (for feature requests)

We don't have monitoring and the status checks has to be performed manually.

bdorney, 2018-02-13:
For a `vfat_info_xhal.py` style tool which reports VFAT3 information please see the proposed register list [here](http://cmsonline.cern.ch/cms-elog/1026255).

For monitoring the communication between CTP7, OH FPGA, VFAT3s, GTBs etc... please see the discussion on [slides 9-13](https://indico.cern.ch/event/694610/contributions/2848959/attachments/1586710/2516541/BDorney_GEMOnlineSW_20180126.pdf).  I think this should be in a dedicated `link_info_xhal.py` script.  Some of this information of course could be repeated in either `amc_info_xhal.py` or `optohybrid_info_xhal.py` tool but a one stop script where it is all displayed is needed in my opinion.

andrewpeck, 2018-02-13:
This is good. There are several registers in the OH that can and should be monitored. I'll compile a list of them here. Implementation in software should probably be done by somebody else (unless you can provide an example script of the exact syntax you want) but for now I'll just compile the list.

mexanick, 2018-02-13:
Thanks guys! I'll implement the things in the software, but I need the inputs you've provided and will provide.

andrewpeck, 2018-02-20:
| Register                                                 |          Expected | Notes                                                                                                                                                           |
|----------------------------------------------------------+-------------------+-----------------------------------------------------------------------------------------------------------------------------------------------------------------|
| GEM_AMC.OH.OH0.TRIG.CTRL.SBIT_SOT_READY                  |        0x00ffffff | 24 bit VFAT mask; Indicates that the trigger Start-of-Transmission frame is good.                                                                               |
| GEM_AMC.OH.OH0.TRIG.CTRL.SBIT_SOT_UNSTABLE               |        0x00000000 | 24 bit VFAT mask; Indicates that the trigger Start-of-Transmission frame has been unstable (i.e. it locked on but has gone bad during running)                  |
| GEM_AMC.OH.OH0.GBT.TX.TX_READY                           |               0x1 | GBT Tx Ready                                                                                                                                                    |
| GEM_AMC.OH.OH0.GBT.RX.RX_READY                           |               0x1 | GBT Rx Ready                                                                                                                                                    |
| GEM_AMC.OH.OH0.GBT.RX.RX_VALID                           |               0x1 | GBT Rx Valid                                                                                                                                                    |
| GEM_AMC.OH.OH0.GBT.RX.CNT_LINK_ERR                       |               0x0 | GBT Link Had Errors -- can be non-zero after startup; ttc resync should reset the counter and it should remain stable                                           |
| GEM_AMC.OH.OH0.ADC.CTRL.CNT_OVERTEMP                     |               0x0 | Number of times ADC detected overtemperature                                                                                                                    |
| GEM_AMC.OH.OH0.ADC.CTRL.CNT_VCCAUX_ALARM                 |               0x0 | Number of times ADC detected overvoltage VCCAUX                                                                                                                 |
| GEM_AMC.OH.OH0.ADC.CTRL.CNT_VCCINT_ALARM                 |               0x0 | Number of times ADC detected overvoltage VCCINT                                                                                                                 |
| GEM_AMC.OH.OH0.CONTROL.RELEASE.DATE                      |                   | Should match expected firmware date                                                                                                                             |
| GEM_AMC.OH.OH0.CONTROL.RELEASE.VERSION.MAJOR             |                   | Should match expected firmware                                                                                                                                  |
| GEM_AMC.OH.OH0.CONTROL.RELEASE.VERSION.MINOR             |                   | Should match expected firmware                                                                                                                                  |
| GEM_AMC.OH.OH0.CONTROL.RELEASE.VERSION.BUILD             |                   | Should match expected firmware                                                                                                                                  |
| GEM_AMC.OH.OH0.CONTROL.RELEASE.VERSION.GENERATION        |                   | Should match expected firmware                                                                                                                                  |
| GEM_AMC.OH.OH0.CONTROL.SEM.CNT_SEM_CRITICAL              |               0x0 | Should be zero, indicating a serious SEU related logic failure; hard reset is requested if non-zero                                                             |
| GEM_AMC.OH.OH0.CONTROL.SEM.CNT_SEM_CORRECTION            |                   | Should be small, but non-zero values are OK on CMS                                                                                                              |
| GEM_AMC.OH.OH0.TRIG.CTRL.SOT_INVERT                      | 0x7992a6/0x799286 | Should be 0x7992a6 on v3a optohybrid, Should be 0x799286 on v3b optohybrid; This needs to be written into the configuration; Current firmware is v3a by default |
| GEM_AMC.OH.OH0.GBT.TX.CNT_RESPONSE_SENT                  |                   | Responses sent should be equal to requests received; they may be different initially but a ttc resync should resolve this                                       |
| GEM_AMC.OH.OH0.GBT.RX.CNT_REQUEST_RECEIVED               |                   | Responses sent should be equal to requests received; they may be different initially but a ttc resync should resolve this                                       |
| GEM_AMC.OH.OH0.CLOCKING.CLOCKING.GBT_MMCM_LOCKED         |               0x1 | Should always be 1                                                                                                                                              |
[... cut ...]
| GEM_AMC.OH.OH0.CLOCKING.CLOCKING.GBT_MMCM_UNLOCKED_CNT   |               0x0 | Should always be 0                                                                                                                                              |
| GEM_AMC.OH.OH0.CLOCKING.CLOCKING.LOGIC_MMCM_UNLOCKED_CNT |               0x0 | Should always be 0                                                                                                                                              |

You may also want to read the S-bit rate counters and see that they are under some threshold (otherwise there is likely some kind of issue on that VFAT)

closed 2018-04-16 by mexanick