cms-gem-daq-project/ctp7_modules #24: [Possible] Bug Report: Differences between Parallel and Series SBIT Rate Scans

bdorney, 2018-02-16:
## Brief summary of issue

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

The two scans should give the same value of rate vs. scan register (bandwidth considerations not accounted for).

## Current Behavior

As shown:

http://cmsonline.cern.ch/cms-elog/1028798

The series and the parallel sbit scans produce different results

### Steps to Reproduce (for bugs)

1. Execute: `sbitThreshScanParallel.py --shelf=3 -s5 -g0 --vfatmask=0x30 --scanmin=0 --scanmax=256 --stepSize=1`
2. Execute: `sbitThreshScanSeries.py --shelf=3 -s5 -g0 --vfatmask=0xfffffe --scanmin=0 --scanmax=254 --stepSize=1 --time=1000`
3. Analyze the two differtent outputs and compare `VFAT0`

## Possible Solution (for bugs)

The case is the following:

1. There is a bug in the CTP7 FW v3.3.6,
2. There is a bug in the OH FW v3.0.9.A, or
3. There is a bug in `sbitRateScanParallelLocal(...)` or the corresponding python script `sbitThreshScanParallel.py`

I will investigate item 3 and update the community.

## Context (for feature requests)

Wrong set point

## Your Environment

* Version used: `2236dc2`
* Shell used: `/bin/zsh`

bdorney, 2018-03-12:
The high peak at high `CFG_THR_ARM_DAC` shown http://cmsonline.cern.ch/cms-elog/1028798 is due to rollover in the register and is expected if one scans beyond the range.

e.g. the `CFG_THR_ARM_DAC` register is `0` to `0xff` (255).  When scanning to 256 this is interpreted by the front end as `0x1` so **noise is high here as expected**.

However this does not describe the difference in shape between the two distributions.  Further investigation is needed.

andrewpeck, 2018-10-04:
Agreed-- the bug was fixed in OH commit: https://github.com/cms-gem-daq-project/OptoHybridv3/commit/0552a74beffc5a16a464691ee090c16b680a8867#diff-f54fdf4a686a7b2e4ccf6e785f9c40d0