cms-gem-daq-project/ctp7_modules #28: Feature Request: Makefile appends library names with identifier

bdorney, 2018-03-08:
## Brief summary of issue

Right now from the libraries that are built it is not possible to see what version/release/commit/branch they were built from.  Suggest that the makefile builds these libraries and appends the name; for example instead of:

```
calibration_routines.so
```

We have

```
calibration_routines_<identifier>.so
```

Where `identifier` is either the release number (e.g. `v3.0.0`), the branch name, the short git commit hash, or some other string that could be used.

Then on the CTP7 in:

```
/mnt/persistent/gemdaq/rpcmodules
```

We have a link:

```
calibration_routines.so -> calibration_routines_v3.0.0.so
```

For stable systems (e.g. P5) we may prefer the `identifier` to be the release number; but for developer systems it may be more useful for it to be the shortened git commit hash.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

To be able to tell what version the libraries.

## Current Behavior

The only way to know what version the libraries are on a given CTP7 is to compile them on your host machine and push them to the card.

## Context (for feature requests)

Long term maintenance.

## Your Environment

* Version used: `develop`
* Shell used: `zsh`

jsturdy, 2018-03-08:
This will depend on how the library is built, i.e., the `soname`
* https://stackoverflow.com/questions/12637841/what-is-the-soname-option-for-building-shared-libraries-for#14613602
* https://en.wikipedia.org/wiki/Soname

In principle, this will matter only if multiple versions of the library are expected to be present at the same time

bdorney, 2018-03-08:
> In principle, this will matter only if multiple versions of the library are expected to be present at the same time

I think that's not the only reason.  For example the question @mexanick asked this morning was "what version of the libraries is on the card."  Probably bad booking keeping on my part but I did not know how to tell from the library file.

jsturdy, 2019-05-06:
Closing by #114

closed 2019-05-06 by jsturdy