cms-gem-daq-project/ctp7_modules #30: Change of Optohybrid FPGA register names in updated parser

andrewpeck, 2018-03-20:
## Brief summary of issue

After the address table was converted from hardcoded values copied into the gem_amc_top.xml, the nodes to access optohybrid registers changed and requires matching updates in any scripts accessing OH registers. 

The CTP7 address table still maintains a hardcoded copy of an old version of the OH register addresses, and the current scripts are still reading out from these old addresses.

### Types of issue

- [X ] Bug report (report an issue with the code)

## Expected Behavior

It should read from the correct add

## Current Behavior

The scripts still read from the old addresses, e.g. ```GEM_AMC.OH.OH0.TRIG.CNT.VFAT0_SBITS ```, while it should be reading from the new addresses, e.g. ```GEM_AMC.OH.OH0.FPGA.TRIG.CNT.VFAT0_SBITS```

### Steps to Reproduce (for bugs)

1. Read an address

## Possible Solution (for bugs)

For all register accesses coming from the Optohybrid, one needs to add "FPGA" as a node, e.g. ```GEM_AMC.OH.OH0.FPGA``` instead of just ```GEM_AMC.OH.OH0```

A not comprehensive list of accesses requiring updates: 

https://github.com/cms-gem-daq-project/ctp7_modules/blob/e082a5d78a52d147ab7f94bde56ccec3a6dc4dd3/src/calibration_routines.cpp#L665-L706

https://github.com/cms-gem-daq-project/ctp7_modules/blob/e082a5d78a52d147ab7f94bde56ccec3a6dc4dd3/src/calibration_routines.cpp#L578

Updates may be required in other places also, e.g. in the non-version-controlled s-bit scan scripts on the CTP7.

andrewpeck, 2018-03-20:
Also, we really should remove the old registers from the XML file so that the software will complain of missing node instead of happily parsing at the wrong address

mexanick, 2018-03-22:
@apeck Please report the address table bug to the appropriate repository. Check with @evka85 which one it should be.

closed 2018-04-16 by mexanick