cms-gem-daq-project/ctp7_modules #32: Bug Report: Error on Generating Documentation via Doxygen/pdflatex

bdorney, 2018-04-17:
## Brief summary of issue

Ran into an error when trying to generate a pdf describing the repo using `doxygen` and `pdflatex`.  Probably missing something? If not then there's a bug here.  Probable need some instructions on how to do this posted somewhere. 

### Types of issue

- [X] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

Attempting to generate the documentation should not crash.

## Current Behavior

Attempting to generate the documentation causes a crash.

### Steps to Reproduce (for bugs)

```
cd $CTP7_MOD_ROOT
doxygen doc/ctp7_modules.doxygen
cd doc/latex
make
```

Generates the following error:

```
% make
rm -f *.ps *.dvi *.aux *.toc *.idx *.ind *.ilg *.log *.out *.brf *.blg *.bbl refman.pdf
pdflatex refman
This is pdfTeX, Version 3.1415926-2.5-1.40.14 (TeX Live 2013)
 restricted \write18 enabled.
entering extended mode
(./refman.tex
LaTeX2e <2011/06/27>
Babel <v3.8m> and hyphenation patterns for english, dumylang, nohyphenation, lo
aded.
(/usr/share/texlive/texmf-dist/tex/latex/base/book.cls
Document Class: book 2007/10/19 v1.4h Standard LaTeX document class
(/usr/share/texlive/texmf-dist/tex/latex/base/bk10.clo))
(/usr/share/texlive/texmf-dist/tex/latex/tools/calc.sty) (./doxygen.sty
(/usr/share/texlive/texmf-dist/tex/latex/base/alltt.sty)
(/usr/share/texlive/texmf-dist/tex/latex/tools/array.sty)
(/usr/share/texlive/texmf-dist/tex/latex/float/float.sty)
(/usr/share/texlive/texmf-dist/tex/latex/base/ifthen.sty)
(/usr/share/texlive/texmf-dist/tex/latex/tools/verbatim.sty)
(/usr/share/texlive/texmf-dist/tex/latex/xcolor/xcolor.sty
(/usr/share/texlive/texmf-dist/tex/latex/latexconfig/color.cfg)
(/usr/share/texlive/texmf-dist/tex/latex/pdftex-def/pdftex.def
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/infwarerr.sty)
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/ltxcmds.sty))
(/usr/share/texlive/texmf-dist/tex/latex/colortbl/colortbl.sty))

! LaTeX Error: File `xtab.sty' not found.

Type X to quit or <RETURN> to proceed,
or enter new name. (Default extension: sty)
```

Seems the file `xtab.sty` is not found.  So probably some env variable is not correctly setup.  Perhaps this should be included in either `$CTP7_MOD_ROOT/setup.sh` or a dedicated script for building the documentation.

## Context (for feature requests)

Trying to read the documentation after re-factoring to understand how the API has changed (e.g. `writeReg(...)` seems to take different argument ordering).

## Your Environment

* Version used: `develop`
* Shell used: `zsh`

mexanick, 2018-04-17:
This is not a repository-related problem, is a shortage in default TeX installation on the CC7. @jsturdy can you follow up please?

bdorney, 2018-04-17:
Probably related comment then, we should probably develop a `requirements.txt` style of file (not just for python/pip) for this, and all other repo's so the system installation procedure catches this (via puppi?).

As discussed: https://github.com/cms-gem-daq-project/xhal/issues/57

closed 2018-04-18 by mexanick