cms-gem-daq-project/ctp7_modules #49: Improvement: multi link CTP7 modules should take NOH as an argument

bdorney, 2018-09-28:
## Brief summary of issue

Right now the most of the multi link rpc modules have hardcoded the number of OH's to be 12 and rely on the ohMask to skip links (or to not consider additional links).

Due to the different link counts for the GE2/1 OH it may be that we have a different number of OH's on the CTP7.  As a result we should not hard code the number of OH's but instead pass this as a parameter. 

This would be a simple task for a new comer.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

Should *not* be a breaking change, just in header files `NOH` should default to 12 for GE1/1.

## Expected Behavior

The RPC request for each of the following functions should have a key `NOH` which determines the number of optohybrids to be looped over in the OH loop:

https://github.com/cms-gem-daq-project/ctp7_modules/blob/a87d951df725e3477ae6f13fd8f1aeb7480c150d/src/calibration_routines.cpp#L1517
https://github.com/cms-gem-daq-project/ctp7_modules/blob/a87d951df725e3477ae6f13fd8f1aeb7480c150d/src/amc.cpp#L477
https://github.com/cms-gem-daq-project/ctp7_modules/blob/a87d951df725e3477ae6f13fd8f1aeb7480c150d/src/vfat3.cpp#L102
https://github.com/cms-gem-daq-project/ctp7_modules/blob/a87d951df725e3477ae6f13fd8f1aeb7480c150d/src/vfat3.cpp#L287

For an example of the implementation see how the key `NOH` in the RPC request is obtained in:

https://github.com/cms-gem-daq-project/ctp7_modules/blob/a87d951df725e3477ae6f13fd8f1aeb7480c150d/src/amc.cpp#L336-L349 

Then this should be used to control the OH loop.

After making your changes you need to try to compile, you can do this by executing:

```bash
export LD_LIBRARY_PATH=/opt/xdaq/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/opt/wiscrpcsvc/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/opt/rwreg/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/opt/xhal/lib:$LD_LIBRARY_PATH
export PETA_STAGE=/data/bigdisk/sw/peta-stage/
source /data/bigdisk/sw/Xilinx/SDK/2016.2/settings64.sh
export XHAL_ROOT=$BUILD_HOME/xhal/
cd ctp7_modules/
source setup.sh
make clean && make
```

Then you'll need to request to test your changes on a given CTP7.

## Current Behavior

Right now the number of OH's is hard coded in the multi link RPC modules, see example `configureVFAT3DacMonitorMultiLink()`:

https://github.com/cms-gem-daq-project/ctp7_modules/blob/a87d951df725e3477ae6f13fd8f1aeb7480c150d/src/vfat3.cpp#L117-L125

## Context (for feature requests)

Due to different number of links for a GE2/1 OH we may have a different number of OH's on the backend, so hardcoding 12 is not a good idea.

jsturdy, 2018-09-28:
Why not just have the module ask the FW how many links it supports?
Or is this to attempt to control the looping at a finer grained level?

The code duplication argument pretty much goes out the window when discussing RPC philosophy (as each module is in effect a standalone operation, rather than part of an object), unless you have the module call another module on the card

bdorney, 2018-09-28:
> Why not just have the module ask the FW how many links it supports?

I didn't realise there was an AMC register that supports this.  Yes this is an ideal solution.

> The code duplication argument pretty much goes out the window when discussing RPC philosophy...unless you have the module call another module on the card

Yes we do, see for example the proposal in issue #48.  Also functions in `vfat.so` and `optohybrid.so` call local functions in `amc.so`.  Also functions in `calibration_routines.so` call local functions from the previous three mentioned shared libraries.  This was done to decrease the number of RPC messages that are sent thus increasing speed of execution (e.g. there's no longer a reason for the RPC request to send the `vfatmask`, this can now be determined by "discovery" on the card, some refactoring is needed for all functions to make use of this).

bdorney, 2018-09-28:
> Why not just have the module ask the FW how many links it supports?

@jsturdy Do you have a copy of the CTP7 documentation?  I tried the link that @andrewpeck sent but now it generates a 404 error and I didn't think to download it at the time.

bdorney, 2018-10-09:
Closed by #51

closed 2018-10-09 by bdorney