cms-gem-daq-project/ctp7_modules #52: Feature Request: Local & Non-local functions for TTC_OVERRIDE_MASK

bdorney, 2018-10-02:
## Brief summary of issue

After discussing with @mexanick we where thinking about having an RPC method for manipulating the `TTC_OVERRIDE_MASK`. 

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

Should have both local and non-local versions, although I expect the primary usage case would be the local version being called by some other `rpcmodule` on the card.

```C++
setTtcOverRideMask(uint32_t override){
    check validity of override, e.g. allowed values {0x1, 0x2, 0x4, 0x8, 0xc}
        - 0xc - triggers TTC hard reset to be sent from TCDS
        - 0x2 - triggers a TTC resync to be sent from TCDS

    check what type of TTC command will be queried from TCDS (e.g. resync or reset)
    get initial value of the counter corresponding to this command type 

    Get address of TTC_OVERRIDE_MASK
    Get initial value of TTC_OVERRIDE_MASK

    while(ttc corresponding counter cmd does not increment){
         set TTC_OVERRIDE_MASK value to ((initial_value & 0xffffff0f) | override )
         set TTC_OVERRIDE_MASK value to initial_value
         read corresponding command counter and determine if it incremented
    }
}

## Current Behavior
<!--- If describing a bug, tell us what happens instead of the expected behavior -->
<!--- If suggesting a change/improvement, explain the difference from current behavior -->
No `rpcmodule` which manipulates TTC override mask

## Context (for feature requests)
<!--- How has this issue affected you? What are you trying to accomplish? -->
<!--- Providing context helps us come up with a solution that is most useful in the real world -->
For automatic link recovery to cause CMS to have the least dead time possible would have register access from the CTP7.  While this might increase GEM dead time slight (due to one or more RPC messages depending on implementation) it will be the fastest solution for CMS.

<!--- Template thanks to https://www.talater.com/open-source-templates/#/page/98 -->

jsturdy, 2018-10-03:
Implementing this in software is not really a long term good idea.
1) This is being proposed to address a very specific problem that doesn't have long reaching applications
1) Setting the override mask is not really the thing that the tool needs to do, rather asking for some `TTC` command to be sent from upstream
1) These `TTC` commands are really only controlling behaviour of the FW anyway

If we want to request a specific command based on a forced `TTS` state, this should really be completely controlled by the firmware (especially if there is *any* concern about deadtime)
The interface with the FW should be simply: ask for `X` `TTC` command (as a part of a manual/automatic script, or in future, as part of a `FixSoftErrorDetected` action)
The FW should then force the necessary `TTS` state, receive the `TTC` command, clear whatever mechanism it used to request it, and continue as though it had received this command normally

bdorney, 2018-10-03:
@jsturdy this seems counter to what we discussed on Mattermost.  The discussion centered around investigating how long each of three methods would leave the system in OOS:

1. Sending a TTC_OVERRIDE to trigger resync with one off uhal commands,
2. As #1 but with uhal dispatch,
3. Using the TTC_OVERRIDE to trigger a resync as a dedicated RPC module

Where number 3 above was the recommendation by @evka85 and is the idea behind the `ask_for_resync.py` script.

So not really clear now how you would like this investigated.

bdorney, 2018-10-03:
Moreover for debugging purposes in the future I don't see why we wouldn't want the ability to trigger a given TTC command to be sent to our back-end electronics.

jsturdy, 2018-10-03:
> So not really clear now how you would like this investigated.

The only thing I was suggesting before was doing 2 from your list, and seeing what the downtime cost actually is (because implementing this in a script is trivial), if the downtime cost is negligible, no strong reason to try for anything more cumbersome at present, otherwise, yes, we have alternative mechanisms to achieve the same stopgaps

> Moreover for debugging purposes in the future I don't see why we wouldn't want the ability to trigger a given TTC command to be sent to our back-end electronics.
> > The interface with the FW should be simply: ask for `X` `TTC` command (as a part of a manual/automatic script, or in future, as part of a `FixSoftErrorDetected` action)

Primary point being, the software shouldn't be using this hack of forcing the `TTS` state to request a `TTC` command, it should simply be able to tell the firmware "hey, ask for `X` `TTC` command!"

bdorney, 2018-10-03:
> The only thing I was suggesting before was doing 2 from your list, and seeing what the downtime cost actually is (because implementing this in a script is trivial), if the downtime cost is [...]

As I discussed on Mattermost when sending the request via `uhal` a TTC resync was *not* sent back, and the recovery procedure hanged indefinitely (since it was waiting for the TTC Resync counter to increment).  So again this can be investigated, but since the register access from the CTP7 will be faster no matter way (keeping the dead time for CMS as low as possible) I think an rpcmodule is still the way to go here.  Yes CMS GEMs incur a small time penalty for the TCP/IP transaction but CMS overall incurs the least penalty in this way.

> Primary point being, the software shouldn't be using this hack of forcing the TTS state to request a TTC command, it should simply be able to tell the firmware "hey, ask for X TTC command!"

I see! Yes this makes a lot of sense.  However right now no such mechanism exists in the FW.  If you would like to wait we can ask @evka85 to incorporate a "REQUEST_TTC_CMD_X" and then the recovery procedure can just write to this register with whatever protocol is desired (either uhal or xhal/rpc).  In which case the dead time for both GEMs and CMS will probably be minimised.

jsturdy, 2018-10-03:
Was your `uhal` test using the register name as in this issue title?  Because that would explain the failure ;-), as in 904 I get:
```bash
Running checks
Executing main
Creating an AMC connection
Initial value of GEM_AMC.DAQ.CONTROL.TTS_OVERRIDE is 0x00000000
Testing where all transactions are dispatched together
Single call took 0.00106906890869
Comparison took 0.247258901596
initial value 0x0, final value 0x0
After reset, final value 0x0
Ran 1000 total checks on GEM_AMC.DAQ.CONTROL.TTS_OVERRIDE, and had 0 mismatches

Testing where each transaction is dispatched individually
Single call took 0.00108003616333
Comparison took 0.959959983826
initial value 0x0, final value 0x0
After reset, final value 0x0
Ran 1000 total checks on GEM_AMC.DAQ.CONTROL.TTS_OVERRIDE, and had 0 mismatches
```

jsturdy, 2018-10-03:
Ah, OK, nope, I just ran a read-only test :-D
```bash
Running checks
Executing main
Creating an AMC connection
Initial value of GEM_AMC.DAQ.CONTROL.TTS_OVERRIDE is 0x00000000
Testing where all transactions are dispatched together
Single call took 0.00103497505188
Comparison took 0.268967151642
initial value 0x0, final value 0xe
After reset, final value 0x0
Ran 1000 total checks on GEM_AMC.DAQ.CONTROL.TTS_OVERRIDE, and had 0 mismatches

Testing where each transaction is dispatched individually
Single call took 0.000864028930664
Comparison took 1.97213387489
initial value 0x0, final value 0xa
After reset, final value 0x0
Ran 1000 total checks on GEM_AMC.DAQ.CONTROL.TTS_OVERRIDE, and had 0 mismatches
```

bdorney, 2018-10-03:
I don't understand what you are trying to illustrate from this command output.

jsturdy, 2018-10-03:
> I don't understand what you are trying to illustrate from this command output.

1) the title of this issue, if as you intended, is incorrect
  * `TTC_OVERRIDE_MASK` does not exist
  * `TTS_OVERRIDE` does
2) `uhal` has *no* problem writing to this register (whether through single or multiple dispatch)
3) apart from writing to this register, the software has *no* say in what else happens, so if a `Resync` is not seen **and** the register was correctly set, the issue is somewhere else

bdorney, 2018-10-03:
I was using:

```python
ttcOutOfSyncValue = 0x2
writeRegister(amcBoard,"GEM_AMC.DAQ.CONTROL.TTS_OVERRIDE",ttcOutOfSyncValue)
```

So the node name matches above.  The OOS value I was writing I understood to be correct.

jsturdy, 2018-10-03:
> There's no real concern about the deadtime at the moment, but we want to reasonably minimize the time required for the recovery. Implementing the recovery as a single RPC call will mask the implementation details (the caller is agnostic whether it is FW setting TTC override or a remote method) and will make it reasonably quick. The plan is to have a script calling one RPC call at the beginning, and if the procedure shows good and stable behavior, we hook it to the `gemsupervisor`

If there is no real concern about increased deadtime, *and* the full procedure is not being implemented as an RPC module(s) (which it actually **can't**, due to the current configuration mechanism), this mix of methodologies is counter-productive
* A hook into the xdaq code, which is increasingly unlikely to happen as the remaining time is dwindling, would be done onto the maintenance branch of the current code: utilizing calls to python scripts, or porting of those calls into a function in the hw class (as done for the phase alignment), not hooking in an `XHALInterface` and corresponding RPC call.

Less time would be wasted by doing a three line addition to a current script, and figuring out what wasn't done correctly when the initial `uhal` invocation didn't work; this can even be done with the V3 setup at P5 while a global run is ongoing (since the long period not in global yesterday was not utilized)

bdorney, 2018-10-03:
Okay we can investigate the `uhal` approach.

But in the meantime I think we should deploy the *current* recovery method, which I have tested in global with both 60 kHz and 80 kHz of randoms.  This then provide a mechanism for automatic recovery for the DOC or DAQ expert to use the remaining time most efficiently, while we iron out a refined procedure.

bdorney, 2019-02-25:
Closing as irrelevant.

closed 2019-02-25 by bdorney