cms-gem-daq-project/ctp7_modules #55: Feature Request: Remove Hard Coded VFAT Number

bdorney, 2018-10-09:
## Brief summary of issue

Again with looking forward to GE2/1 and ME0, we can re-use many of our rpcmodules for future GEM DAQ hardware.  However we have hardcoded the number of VFATs per optical link.

We need to come up with a way of identifying the number of VFATs per link and then using this mechanism when indexing/looping over VFATs in rpcmodules. 

### Types of issue

- [ ] Bug report (report an issue with the code)
- [X] Feature request (request for change which adds functionality)

## Expected Behavior

I assume the `GEM_AMC.OH_LINKS` block of the FW will be reused (@evka85?).  So here we could implement an rpcmodule (local, or local & non-local) which would try to get the following node:

```
GEM_AMC.OH_LINKS.OHX.VFATY.SYNC_ERR_CNT
```

If the node is found, it increments Y and tries to get the node again, and continues until the node is not found (e.g. Y =24 in GE1/1).  This might require some exception handling.  I don't remember if a node not found exception is raised or not.

Alternatively if there was a register in the AMC FW that specified the number of VFATs per OH then this register could just be read by the module.

Then whenever indexing or looping over VFATs this return value of this rpcmodule should be used as the maximum VFAT number.

## Current Behavior

The number of VFATs is hard coded to 24.

## Context (for feature requests)

Not all GE2/1 modules will have 24 VFATs.

bdorney, 2019-02-04:
This is probably possible now that `hw_constants.h` has been included and the `GEM_VARIANT` field exists in the `Makefile`.

lpetre-ulb, 2020-03-01:
Implemented in https://github.com/cms-gem-daq-project/ctp7_modules/pull/132. Fixed-size return types will be tackle in this issue https://github.com/cms-gem-daq-project/ctp7_modules/issues/161.

Should be closed, but I don't have the rights to do so. BTW, I have vastly different permissions between software repositories; I can force merge in some while I cannot even assign people to issues and PR in others.

closed 2020-03-02 by jsturdy

jsturdy, 2020-03-02:
The scoped permissions were by design, and limited by the previous roles that github had allowed (they have since added a few more roles for better granularity)