cms-gem-daq-project/ctp7_modules #58: Integrate ipbus into repo

jsturdy, 2018-10-11:
## Brief summary of issue
With the migration of the `memhub` library to this repo, the `ipbus` binary should likewise migrate here.

### Types of issue
- [x] Feature request (request for change which adds functionality)

## Context (for feature requests)
The `ipbus` binary should be built with this package, and included in the `RPM` that is installed on the CTP7, along with the `libmemhub.so` library

mexanick, 2018-10-11:
Also improvement of packaging tools is expected along with this migration

bdorney, 2018-10-12:
Isn’t this already built in xhal and migrated to the card with setup_ctp7.sh?

lpetre-ulb, 2020-05-18:
The CTP7 `ipbus` daemon is part of `xhal` which contains its own version of `memhub`. Probably does not deserve a fix in legacy.

closed 2020-05-18 by lpetre-ulb