cms-gem-daq-project/ctp7_modules #62: Maintenance: Change Local CTP7 Modules to Struct's as input arguments (Good for Newcomer)

bdorney, 2018-10-25:
## Brief summary of issue

Several of our `rpcmodule`s have *huge* argument lists that are cumbersome to use and also maintain; @mexanick and I have wanted to trim this down for some time to use several struct's to handle this instead. 

### Types of issue

- [ ] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)
- [x] Maintenance (improves long-term maintainability of the code base)

## Expected Behavior

There are a set of common parameters that are typically passed to the local functions on the CTP7.  These could be stored in a set of `struct`'s and then instances of these `struct`'s could be passed as input.

Goes without saying but the non-local functions which call the local functions mentioned below should *also* be modified accordingly.

### Expected Struct's

```C++
    struct ParamCalPulse{
        bool enable; //true (false) turn on (off) calpulse
        bool isCurrent; //true (false) is current injection (voltage pulse)

        uint32_t duration; //duration in BX's (CFG_CAL_DUR)
        uint32_t extVoltStep; //External voltage step 0->disable; 1->enable (CFG_CAL_EXT)
        uint32_t height; //height of calpulse (CFG_CAL_DAC)
        uint32_t phase; //phase of calpulse (CFG_CAL_PHI)
        uint32_t polarity; //polarity of calpulse 0->pos; 1->neg (CFG_CAL_SEL_POL)
        uint32_t scaleFactor; //current pulse scale factor (CFG_CAL_FS)

        ParamCalPulse(){
            enable = false;
            isCurrent = false;

            duration = 0x1ff;
            extVoltStep = 0x0;
            height = 0x0;                                                                                                                                                                                                                
            phase = 0x0;
            polarity = 0x0;
            scaleFactor = 0x0;
        }
    }; //End ParamCalPulse

    struct ParamScan{
        //Hardware selection
        uint32_t ohN;
        uint32_t ohMask;
        uint32_t vfatN;
        uint32_t vfatMask;
        uint32_t chan;      //channel of interest

        //Params
        bool useUltra;
        bool useExtTrig;

        uint32_t dacMax;    //Maximum dac value
        uint32_t dacMin;    //Minimum dac value
        uint32_t dacSelect; //DAC to use for Monitoring
        uint32_t dacStep;   //step size for dac
        uint32_t nevts;     //Number of events
        uint32_t waitTime;  //unit of time; uints depend on function

        std::string scanReg;     //Register to scan against

        ParamScan(){
            vfatMask = 0x0;
            ohMask = 0xfff;

            useUltra = true;
            useExtTrig = false;

            dacMax=254;
            dacMin=0;
            dacStep=1;
            nevts=100;
        }
    }; //End ParamScan

    struct ParamTtcGen{
        bool enable;

        uint32_t L1Ainterval;
        uint32_t mode;
        uint32_t nPulses;
        uint32_t pulseDelay;
        uint32_t pulseRate;
        uint32_t type;

        ParamTtcGen(){
            enable = false;

            L1Ainterval = 250;
            pulseDelay = 40;
            pulseRate = 40079000 / L1Ainterval;
        }

        uint32_t calcRate(){
            if(L1Ainterval > 0){
                pulseRate = 40079000 / L1Ainterval;
            }
            else{
                pulseRate = 0;
            }

            return pulseRate;
        }
    }; //End ParamTtcGen
```

These `struct`s respectively provide info for:

 - Configuring the calibration pulse between voltage step pulsing or current injection (see VFAT3 manual),
- The hardware to be scanned,
- The TTC Generator parameters

This could be defined in `include/utils.h`.  That seems appropriate.

### Functions of Interest

Functions in `optohybrid.*` and `calibration_routines.*`.

From `optohybrid.*` we have:

- `configureScanModuleLocal(...)`, and
- `getUltraScanResultsLocal(...)`

However we could drop support for these entirely and remove them from the repo (I would be in favor of this...but perhaps this deserves it's own issue...).

From `calibration_routines.*` we have:

- `confCalPulseLocal(...)`,
- `ttcGenConfLocal(...)`,
- `genScanLocal(...)`,
- `sbitRateScanLocal(...)`,
- `sbitRateScanParallelLocal(...)`,
- `checkSbitMappingWithCalPulseLocal(...)`, and
- `checkSbitRateWithCalPulseLocal(...)`.

### Usage Example 

So the present version of `genScanLocal(...)` looks like:

https://github.com/cms-gem-daq-project/ctp7_modules/blob/4d82a717400fd96cb6f18104b58b93a1e8751036/src/calibration_routines.cpp#L415

But it could look like:

```C++
void genScanLocal(localArgs *la, uint32_t *outData, ParamCalPulse *calParams, ParamScan *scanParams)
```

Which is much simpler.

## Current Behavior

Long laundry list of parameters passed to local functions.

We should learn from the use of the `localArgs` struct.

## Context (for feature requests)

Maintainability of codebase.

bdorney, 2018-10-25:
Think this could be a good potential task for @AndrewLevin.

AndrewLevin, 2018-10-25:
ok, I will do it

bdorney, 2018-10-25:
`dacScanLocal(...)` may also benefit from using `ParamScan *scanParams` as an argument.

bdorney, 2018-10-28:
Yes indeed it may not be so clear, but look at how this value is used:

https://github.com/cms-gem-daq-project/ctp7_modules/blob/4d82a717400fd96cb6f18104b58b93a1e8751036/src/calibration_routines.cpp#L750-L754

and

https://github.com/cms-gem-daq-project/ctp7_modules/blob/4d82a717400fd96cb6f18104b58b93a1e8751036/src/calibration_routines.cpp#L773-L774

This is being written to the `GEM_AMC.OH.OHX.FPGA.TRIG.CTRL.VFAT_MASK` register.  It's masking VFATs from the trigger block in the OH FW.

bdorney, 2018-10-28:
> so, maybe a better name is ohVfatMask?

For now let's leave it as is.  Then open a separate issue for changing this since right now if you change the key name in the non-local version of the function it would be a breaking change and require a separate PR in the legacy branch of `xhal`.  That branch will be depreciated.

In the future `xhal` will only be responsible for atomic transactions and provide an interface class that will be used by classes in `cmsgemos`.  Once changes to `cmsgemos` are ready we can rename this key and have the corresponding change in `cmsgemos`.

lpetre-ulb, 2020-05-18:
Could probably be implemented in the new repository. Will see how things move forward.

closed 2020-05-18 by lpetre-ulb