cms-gem-daq-project/ctp7_modules #65: [feature request] Phase shifting module

jsturdy, 2018-11-09:
## Brief summary of issue
To remove the phase shifting from the `cmsgemos` implementation, and away from a script living on the CTP7, this should be incorporated into a module (until it is fully implemented in the FW)

### Types of issue
- [x] Feature request (request for change which adds functionality)

## Possible Solution (for bugs)
Look at (`python`):
https://github.com/jsturdy/GEM_AMC/blob/fe3780048191b6a23ef04d1d430d3b53c5e8e8b8/scripts/test_phase_shifting.py#L99-L451 

for the current incarnation (and associated helper functions), which *should* be functionally identical to the `c++` implementation done here (with `uhal`):
https://github.com/jsturdy/cmsgemos/blob/1e8e6a90181eb7c4b4bf4cf495fa4f3437b81d4f/gemhardware/src/common/HwGenericAMC.cc#L549-L1019

bdorney, 2018-11-13:
I'll take this one on.  This is a purely AMC function correct?

jsturdy, 2018-11-13:
> This is a purely AMC function correct?

Yes, N.B., it **only** works with V2 FW (unless @evka85 has ported the updated TTC module to the V3)

closed 2019-02-25 by bdorney