cms-gem-daq-project/ctp7_modules #67: Feature Request: VFAT Temperature Monitoring in daq_monitor

bdorney, 2018-11-12:
## Brief summary of issue

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

User will supply the following information in the rpc message (i.e. rpc keys):

- Which OH's to scan `ohMask` (12 bit number, if 1 in the N^th bit use the N^th OH, otherwise skip).
- Which ADC to use via a boolean `useExtRefADC` if false (true) use ADC0 (ADC1)
- Which temperature sensor to use via a boolean `useIntTemp` if true (false) use internal (external) temperature sensor.

The basic outline should be:

```C++
void getmonVFAT3Temp(RPC Msg, RPC Response){
   Get info from RPC message

   determine dacSelect, this will be 37 (38) if useIntTemp is true (false)

   Loop over NOH's
       If the OH is masked, skip
       get the vfatmask from getOHVFATMaskLocal()
       configure the DAC monitoring with configureVFAT3DacMonitorLocal()
       read VFAT3 ADC with readVFAT3ADCLocal()
       store read back info in output RPC response
}
```

Compare with existing examples in `daq_monitor` to make sure all the bell's and whistles (specifically for how to get `ohMask` and `NOH` keys).

- [getOHVFATMaskLocal](https://github.com/cms-gem-daq-project/ctp7_modules/blob/ab9f57487ab79e3f37710da4db84835d80b1401f/src/amc.cpp#L42)
- [configureVFAT3DacMonitorLocal](https://github.com/cms-gem-daq-project/ctp7_modules/blob/ab9f57487ab79e3f37710da4db84835d80b1401f/src/vfat3.cpp#L50)
- [readVFAT3ADCLocal](https://github.com/cms-gem-daq-project/ctp7_modules/blob/ab9f57487ab79e3f37710da4db84835d80b1401f/src/vfat3.cpp#L262)

You could test this then by modifying `monitorTemperatures.py` to use your RPC call instead of the set of calls made presently.  Or ask @mexanick or @jsturdy for help on incorporating this into `cmsgemos`.

## Current Behavior

There's no single entry function for convenient temperature monitoring.

## Context (for feature requests)

Need a way to monitor temperatures of VFAT3 in P5 and especially on QC8 to ensure the cooling is working well and cooling plate is making good contact.

bdorney, 2018-11-13:
In case it wasn't clear from the above, in this case there is no need for a "local" function, the remote version calls a set of pre-existing local functions that do the job.

lpetre-ulb, 2020-05-18:
Moved to GitLab ([cmsgemos#92](https://gitlab.cern.ch/cmsgemonline/cmsgemos/-/issues/92)).

closed 2020-05-18 by lpetre-ulb