cms-gem-daq-project/ctp7_modules #68: Discussion: Implementation of Namespaces?

bdorney, 2018-11-13:
## Brief summary of issue

As we are increasing our CTP7 modules we should probably start to think about namespaces (or more aptly we should have thought about this from the beginning...?).

Right now the libraries include a lot of each and it might be time to further compartmentalise things.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)
- [X] Discussion

## Expected Behavior

Having namespaces something like:

- `amc::` for `include/amc.h`
- `vfat3::` for `include/vfat3.h`,
- etc...

Maybe this is overkill?

## Current Behavior

No namespaces are used in this repo.

## Context (for feature requests)

Longer term maintenance and preventing compile issues from say global variable declaration.

For example relating to https://github.com/cms-gem-daq-project/ctp7_modules/issues/65.  If we copy/paste necessary functions from `cmsgemos` several constants are declared in multiple of the required functions, namely:

```C++
  const int PHASE_CHECK_AVERAGE_CNT = 100;                                                                                                                                                                                                   
  const int PLL_LOCK_READ_ATTEMPTS  = 10;
  const double PLL_LOCK_WAIT_TIME   = 0.00001; // wait 100us to allow the PLL to lock
```

Perhaps they should be declared "globally" in `include/amc.h` and we introduce a namespace `amc::` but I'm not sure the best structure to use (e.g. ideally for maintainability we would want a similarly or at least logical structure that's used in `cmsgemos` or simply a structure that follows from the filename in question...?).

jsturdy, 2018-11-15:
@mexanick should probably comment on the restrictions with the way this code has to be written to accommodate pure `C` stuff vs. OO `C++` stuff (I vaguely recall some discussion of the finer points of this in the past but definitely not enough to )

mexanick, 2018-11-16:
Well, the restriction is in the way how the module functions are exposed to module manager. So in principle namespaces eventually can be used, but I feel like it is an overkill and our project can be written in C.

lpetre-ulb, 2020-05-18:
Naturally implemented with the templated RPC methods.

closed 2020-05-18 by lpetre-ulb