cms-gem-daq-project/ctp7_modules #72: Move the GBT tools from reg_utils python scripts to RPC module

lpetre-ulb, 2018-11-26:
## Brief summary of issue

As described in [this issue](https://github.com/cms-gem-daq-project/reg_utils/issues/41), the `gbt.py` tool cannot be called on the CTP7 with the 12 OH's address table, because of out-of-memory error.

The best solution would be to implement the features of the [`gbt.py` tool](https://github.com/cms-gem-daq-project/reg_utils/blob/master/python/reg_interface/scripts/gbt.py) within RPC functions.

The required features are 

* Program the GBTX from config file.
* Perform a phase on the GBTX.
* Manually set the phase of one e-link.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

The functions currently supported by the `gbt.py` python tool must be ported to RPC modules.

## Current Behavior

The GBT tool is currently a python script and not a RPC module.

## Context (for feature requests)

The `gbt.py` tool cannot be called on the CTP7 with the 12 OH's address table because of out-of-memory errors.

bdorney, 2018-11-26:
So this should be compartmentalised into a series of functions.

I see the following functions being necessary:

- `writeGBTRegLocal(LocalArgs *la, uint32_t ohN, uint32_t gbtN, ...)`: For OHX this should write a GBTX register with value Y.
- `writeGBTConfigLocal(....)` this should apply a given GBT config to the GBTx on OHX
    - This should have a local and a non-local function
    - I think the non-local function should have an RPC key for binary data that is the GBT config, this can be passed to this function at the DAQ machine side either as an input file or a DB read.
   - The local version of this function obviously calls `writeGBTRegLocal(...)`
- `scanGBTPhaseLocal(...)` this should ask to scan the phase on OHX GBTY and it should take as input a parameter which specifies the number of phase scans that should be performed
   - The local version of this function obviously calls `writeGBTRegLocal(...)`
   - There should be a local and non-local version of this function
   - This could also have a multi link non-local function (see for example `dacScanMultiLink(...)`.
   - For each GBT e-link (VFAT Pos) this should return in the RPC response a key which stores a word array where the array position corresponds to e-link phase and the array contents is the number of times that phase was "good."
- `writeGBTVFATPhaseLocal(...)`: this should write the phase of a given VFAT position to be X
   - The local version of this function obviously calls `writeGBTRegLocal(...)`
   - Sets the phase of a given VFAT to be the value
   - Local and non local version
   - This *could* additionally also have a MultiLink version however if long & short detectors are mixed on the same CTP7 it's likely this multi link version would never be used...still it could be in the API.

This is how I would design this development.

Once you have the `writeGBTRegLocal(...)` function the rest would follow shortly.

bdorney, 2018-11-26:
You should probably solicit feedback from @jsturdy and @mexanick before starting to ensure the design is well thought.

In the absence of a hacked address table just for gbt.py (*BAD* DO NOT DO) we have no way to program GBT's in the 12 OH CTP7 FW without significant tom-foolery.

Critical path item.

bdorney, 2018-11-27:
One further thought, `writeGBTVFATPhaseLocal(...)` should be called by `scanGBTPhaseLocal(...)`.

lpetre-ulb, 2018-11-27:
This was my original plan, but there is a slight implementation difficulty.

Looking at the [GBTX registers tables](https://espace.cern.ch/GBT-Project/GBTX/Manuals/gbtxManual.pdf) (from page 160), the phase registers require masked writes. If I believe the current [gbt.py code](https://github.com/cms-gem-daq-project/reg_utils/blob/7b9cf050c63dae5f3a9b5803389130cff2f039fa/python/reg_interface/arm/gbt_utils.py#L60-L67) and [firmware code](https://github.com/cms-gem-daq-project/GEM_AMC/blob/v3/common/hdl/slow_control/gbtx_ic_controller.vhd), reading registers from the GBTX is not yet implemented.

So, one cannot do masked writes without knowing the initial value (which is currently read from the configuration file). The RPC keys might be not very clean if we need to transport the initial value for each register to write.

The other solution is to set all the values of one GBTX group at once, either in the manual write or the scan. The remaining issue is linked to the FPGA and SCA phases. I still need to think a little about that.

jsturdy, 2018-11-27:
If you feel that these GBTX registers should be exposed to the SW (rather than simply through the I2C interface), you can [open an issue](https://github.com/evka85/GEM_AMC/issues/new)

On the other hand, nothing prevents you from reading the full register as it currently is (using I2C), modifying the bits necessary (in the RPC module), and writing the updated value back (using I2C) (basically replicating the current behaviour)

lpetre-ulb, 2018-11-28:
Finally, I found that the MSB of the registers which configure the phases are not used in the 320MHz mode. Writing 0's to them should not bring any problem. So the `scanGBTXPhase` and `writeGBTXPhase` can work without knowing the initial value of the MSB's (which is certainly 0x0).

I guess that reading the GBTX registers through the IC channel could be useful at some point, but I think think that is a priority and I don't know if it is worth opening an issue if no one has a need for that feature.

commit c323274d (2019-01-16): Merge pull request #75 from lpetre-ulb/feature/GBTXTools_v2

Closes #72
closed 2019-01-16 by jsturdy