cms-gem-daq-project/ctp7_modules #80: DAC scans with cached registers fail for nOH != 0

lpetre-ulb, 2019-01-17:
## Brief summary of issue

Trying to perform DAC scans with cached registers on multiple links on `eagle64` at first and then on `eagle63` (CTP7 firmware > 3.7.1). The scans fail when OH's different than 0 are unmasked.

### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

I expect the DAC scans to run without error for all the OH's.

## Current Behavior

The DAC scans fail for `nOH != 0`.

Here is the command result :

```
[gemuser@ctp7-pc ~]$ dacScanV3.py eagle63 0x4 --dacSelect 2 -f /tmp/dacScan.root
Open pickled address table if available  /opt/cmsgemos/etc/maps//amc_address_table_top.pickle...
Initializing AMC eagle63
opened connection
Getting CHIP IDs of all VFATs
Scanning DAC: CFG_BIAS_PRE_I_BIT on all links
Caught an error: memsvc error: Invalid address: 0xdeaddead
Traceback (most recent call last):
  File "/opt/cmsgemos/bin/dacScanV3.py", line 281, in <module>
    scanAllLinks(args, calTree, vfatBoard)
  File "/opt/cmsgemos/bin/dacScanV3.py", line 69, in scanAllLinks
    raise Exception('RPC response was non-zero, this inidcates an RPC exception occurred')
Exception: RPC response was non-zero, this inidcates an RPC exception occurred
```

And the log on the CTP7 :

```
Jan 17 18:20:42 eagle63 local0.notice rpcsvc[15590]: rpcsvc: Client connected from 192.168.0.180
Jan 17 18:20:42 eagle63 local0.info rpcsvc[15590]: extras:  Memhub initialized a semaphore. Current semaphore value = 1
Jan 17 18:20:43 eagle63 local0.info rpcsvc[15590]: amc.getOHVFATMaskMultiLink: Determined VFAT Mask for OH2 to be 0x1
Jan 17 18:20:44 eagle63 local0.info rpcsvc[15590]: calibration_routines.dacScanMultiLink: Getting VFAT Mask for OH2
Jan 17 18:20:44 eagle63 local0.info rpcsvc[15590]: calibration_routines.dacScanMultiLink: Performing DAC Scan for OH2
Jan 17 18:20:44 eagle63 local0.info rpcsvc[15590]: calibration_routines.dacScanMultiLink: System release major is 3, v3 electronics behavior
Jan 17 18:20:44 eagle63 local0.info rpcsvc[15590]: calibration_routines.dacScanMultiLink: Scanning DAC: CFG_BIAS_PRE_I_BIT
Jan 17 18:20:44 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: Key: GEM_AMC.OH.OH2.GEB.VFAT1.ADC0 is NOT found
Jan 17 18:20:44 eagle63 local0.info rpcsvc[15590]: calibration_routines.dacScanMultiLink: VFATs not in 0x1 were set to run mode
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0x0000001b
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0xdeaddead
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0x0000001b
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0xdeaddead
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0x0000001b
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0xdeaddead
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0x0000001b
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0xdeaddead
Jan 17 18:20:45 eagle63 local0.err rpcsvc[15590]: calibration_routines.dacScanMultiLink: read memsvc error: Invalid address: 0x0000001b
```

Note that the `GEM_AMC.OH.OH2.GEB.VFAT1.ADC0_CACHED` exists in the LMDB database (checked with a small C++ program).

### Steps to Reproduce (for bugs)

1. Launch the DAC scan with a command similar to the above-mentioned one : `dacScanV3.py <eagle> 0x2 --dacSelect 2 -f /tmp/dacScan.root` on a CTP7 with address tables for 3.7.1 or superior.
2. The DAC scan fails.

## Possible Solution (for bugs)

I traced back the error to these lines :https://github.com/cms-gem-daq-project/ctp7_modules/blob/c323274d9344ebb922500832a53c9a2af11eede4/src/calibration_routines.cpp#L1176-L1178

Indeed, the `assign` method stores a pointer the the raw data of the `std::string` : https://github.com/cms-gem-daq-project/ctp7_modules/blob/c323274d9344ebb922500832a53c9a2af11eede4/include/lmdb_cpp_wrapper.h#L1059-L1064

As `strRegBase+"ADC0_CACHED` is a temporary object, its underlying memory may or may not exist depending on the execution flow, compiler, compiler options, ...

The fix is simple : make the `std::string` passed to `assign` live as long as the `lmdb::val` object.

## Your Environment

* RPC modules : https://github.com/cms-gem-daq-project/ctp7_modules/commit/9911586985520c6ad5c2e6206708ccf7c8f13f24 or https://github.com/cms-gem-daq-project/ctp7_modules/commit/c323274d9344ebb922500832a53c9a2af11eede4 (no changes in the `calibration_routines.so` module)
* CTP7 : `eagle63` or `eagle64`
* CTP7 firmware : at least 3.7.1 (and so with cached registers)

jsturdy, 2019-01-18:
I see, followup questions:
* Are there any other places where similar problems are likely to arise?
* Is the `assign` operator intended to be used in this way?
* Is #81 the right solution to this issue?  I.e., isn't it better to solve this inside the `lmdb` wrapper?
* Would it be better if the `lmdb` wrapper provided a `hasKey` function (does it not already?), rather than the current per-function implementation?

lpetre-ulb, 2019-01-18:
>     * Are there any other places where similar problems are likely to arise?

I don't think there are; the only other file where `assign()` is used is in `utils.cpp` and I don't spot issues there.

>     * Is the `assign` operator intended to be used in this way?

I think it is, but the documentation is rather thin. At least, in the C library, it is.

>     * Is #81 the right solution to this issue?  I.e., isn't it better to solve this inside the `lmdb` wrapper?
>     * Would it be better if the `lmdb` wrapper provided a `hasKey` function (does it not already?), rather than the current per-function implementation?

The following function in the `lmdb` wrapper seems to provide the `hasKay` functionality. I'll give it a try.
```c++
template<typename K> bool lmdb::dbi::get(MDB_txn* const txn, const K& key) const
```

bdorney, 2019-01-21:
I don't really see anywhere that the problem is OH0 being masked, but rather the `assign` operator is doing memory shenanigans on the `std::string` passed to it.  If it is truly a case where `OH0` is missing (And I don't see how it is...) prove with an example, e.g. using OH0 and OHX and then again with just OHX (X != 0).

bdorney, 2019-01-21:
Moreover, to me it seems like the issue is that we are trying to accommodate an older FW version with backwards compatibility (which has raised additional questions in #76 and shown to have this problem).

I don't see why we would want to move a setup from 3.7.Y back to 3.X.Y for X < 7.  Perhaps for this reason we should just drop the backwards compatibility, and from now on use the ADC cached registers both in the dacScan application and the ADC monitoring (in `vfat3.cpp`).

I have a PR ready to go if this is the route we wanted to go but felt like before adding another PR on a closely related topic (#76 and #80) we should discuss it first to avoid confusion.

@lpetre-ulb @jsturdy @mexanick

jsturdy, 2019-01-21:
> I don't see why we would want to move a setup from 3.7.Y back to 3.X.Y for X < 7.

I agree, this will not likely ever happen.  What could happen, is a fix in core FW functionality that reverts the "hack" that is using cached registers.  That would be the circumstance I would envision as a reason to keep these implementations.

jsturdy, 2019-01-21:
This issue (#80) addresses a specific fault of the current software, namely, the undefined behaviour of some already existing code in `calibration_routines.cpp`
As that stands (and begging the question of why the same functionality is currently implemented in two locations), a fix to the code reported here is the correct first step, #76 may be in the scope of reworking how the current routine is done.
As I understand it, #76 is adding similar (the same) functionality in a separate location (`vfat3.cpp`).

@bdorney, do I understand correctly that  #76 is a "V3" only implementation, while this fix was done in a "generic"/"legacy" module (that may now be "V3" only)?  If not, can someone quickly summarize why `readVFAT3ADCLocal` and `dacScanLocal` seemingly reimplement the same functionality?

bdorney, 2019-01-21:
I'm not sure what you mean by "generic"/"legacy" module; all functions described are v3 electronics only.

I agree that it may be possible to update `dacScanLocal` to call `readVFAT3ADCLocal` N times per DAC scan point and then all of these "bells and whistles" could be done inside `readVFAT3ADCLocal`.  Indeed this would be more advantageous especially since `readVFAT3ADCLocal` is used also by the monitoring code.

This was for sure an oversight on my part.

jsturdy, 2019-01-21:
> I agree that it may be possible to update `dacScanLocal` to call `readVFAT3ADCLocal` N times per DAC scan point and then all of these "bells and whistles" could be done inside `readVFAT3ADCLocal`. [...]

OK, so if we accept this issue (#80), assuming an improved implementation of #81, then #76 could be a proper rethink to streamline things, taking into account the comments there?
Or is #76 also a "get it done, functionality needed" PR?

bdorney, 2019-01-21:
> I agree, this will not likely ever happen. What could happen, is a fix in core FW functionality that reverts the "hack" that is using cached registers. That would be the circumstance I would [...]

Do we expect this to be likely?  As I understood this was nontrivial and required a redesign of both our SW/FW and the CTP7 linux image.  Ideally to reduce complexity I would just drop backwards compatibility.  For every ADC read this simply adds one transaction and at most one wait time (~20us).   

For a DAC scan on all links this adds `(12*24*(time of one addition read in us))+20us` per point.  For ADC monitoring (e.g. in P5) this adds this time once.

It would reduce complexity to just drop this and keep the current implementation but migrate it to `readVFAT3ADCLocal` and then `dacScanLocal` just calls that function.

bdorney, 2019-01-21:
> Or is #76 also a "get it done, functionality needed" PR?

I have a private patch running in the gem lab so there is no rush.  Let's choose the right thing for long term and then implement that.

jsturdy, 2019-01-21:
@lpetre-ulb, if

> template<typename K> bool lmdb::dbi::get(MDB_txn* const txn, const K& key) const;

is possible, then the lines (and other locations similar)
https://github.com/cms-gem-daq-project/ctp7_modules/blob/e8f202680ddb4c35e1d87dba1cdb600ef62607c3/src/calibration_routines.cpp#L1163-L1167

Can be replaced with
```c++
            if (la->dbi.get(la->rtxn,"REG_NAME")) {
```
which is much cleaner.

lpetre-ulb, 2019-01-21:
> I don't really see anywhere that the problem is OH0 being masked, but rather the `assign` operator is doing memory shenanigans on the `std::string` passed to it. If it is truly a case where `OH0` is [...]

Indeed, OH0 working was just pure "luck"; the title is of the issue might not be the best, because is based on how the issue was discovered.

> Can be replaced with
> 
> ```c++
>             if (la->dbi.get(la->rtxn,"REG_NAME")) {
> ```
> 
> which is much cleaner.

This is my plan, but the implementation of `template<typename K> bool lmdb::dbi::get(MDB_txn* const txn, const K& key) const;` does not behave as I expected. I'll add a new overload `lmdb::dbi::get` for usage with `std::string`'s. The PR update is coming in a few minutes.

closed 2019-01-28 by jsturdy