cms-gem-daq-project/ctp7_modules #109: [feature request] Implement exception handing in RPC callbacks

jsturdy, 2019-03-28:
## Brief summary of issue
To better handle error conditions and unexpected behaviour, we should implement common exception types, and some handlers for cases where a usual course of action needs to be taken.
The RPC method should always be handling the received exceptions and setting the `error` key in the response message (or handling, if it is appropriate for the method)

### Types of issue
- [x] Feature request (request for change which adds functionality)

## Expected Behavior
* Local methods should *not* set an `error` key in the `RPCMessage` response
* RPC callbacks should set the `error` key in the `RPCMessage` response in the event of an exception or error in the local method

## Current Behavior
Some local methods return error codes, some set the `error` key, and some `throw` an exception.

bdorney, 2019-04-03:
Isn't this a duplicate issue to #85?

bdorney, 2019-04-03:
> Some local methods return error codes, some set the error key, and some throw an exception.

Additionally which methods `throw`?  None come to mind presently.  I'm only aware of those that set the error key.