cms-gem-daq-project/ctp7_modules #116: [Discussion] When to use Slow Control Only Mode?

bdorney, 2019-04-05:
## Brief summary of issue

Due to how the VFAT comm port prioritizes data transmission we should think how and when we should use the slow control only mode (`GEM_AMC.GEM_SYSTEM.VFAT3.SC_ONLY_MODE`).

It is possible, based on testing at QC7, to generate bus errors or even cause a VFAT to go out of sync if high frequency (period O(20us)) slow control commands come into the VFAT while it's in run mode.

This issue hopes to clarify under what conditions `SC_ONLY_MODE` should be used and lead to an appropriate re-factoring.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)
- [x] Discussion

## Your Environment

* Version used:
* Shell used:

bdorney, 2019-04-05:
I don't think @cgalloni is part of the team since I cannot tag them.

jsturdy, 2019-04-05:
A few notes:
1) If the chip is not receiving L1As (and hence formatting data packets) the data transmission prioritization *should* not be an issue (trigger data is from a different port)
1) The manual states wrt the SC "It is therefore possible to start and stop the slow control communication in mid flow and resume when the Comm-port is free", though I am not sure how the CTP7 interacts with this.
1) The manual states that "Reapplication of a ReSync signal may be required if synchronization is lost" and "If register values have ben changed then it is advised (although not always necessary) to reapply a ReSync signal", thought I do not know what the operational implications of not doing this may be....

bdorney, 2019-04-06:
Adding some relevant info here from @evka85:

> Whenever VFAT is sending DAQ data to the CTP7, that will delay the slow control response (any slow control packet that is being sent back to CTP7 will take longer), and if L1A rate is high, or just an unfortunate coincidence of close-by L1As, this can delay the slow control response enough so that CTP7 will timeout on it and report bus error. 
> But in this case I would expect that stopping all L1As would fix the issue (you could use the GEM_AMC.TTC.CTRL.L1A_ENABLE register to block them while you're doing critical slow control commands). Putting the VFAT3 into slow-control-only mode should not be necessary, but of course this can be used instead of blocking L1As.

So perhaps the right thing to do is not to use the slow control only mode but instead to tell the AMC to block L1A's.