cms-gem-daq-project/ctp7_modules #119: Bus error persists after reading disconnected SCA-ADC channels

ram1123, 2019-04-25:
## Brief summary of issue
One we access a disconnected channel then it gives bus error. And after that it starts giving error for the connected channels also.

### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

This shows flowing error:

#### While running the command:
```bash=
[rasharma@gem904daq01 bin]$ ./blaster eagle34 08 0x200
calling amc.readADCCommands on eagle34 with 8 and 512...done!
RPC ERROR:memsvc error: Bus error accessing 0x66c0400c
```

#### In log pannel it shows:
```bash=
Apr 15 10:45:39 CTP7 local0.notice rpcsvc[5827]: rpcsvc: Client connected from 192.168.0.180
Apr 15 10:45:39 CTP7 local0.info rpcsvc[5827]: amc:  Memhub initialized a semaphore. Current semaphore value = 1
Apr 15 10:45:39 CTP7 local0.info rpcsvc[5827]: amc.readADCCommands: DEBUG: 3: Name of ADC Channel to read: 8
Apr 15 10:45:39 CTP7 local0.info rpcsvc[5827]: amc.readADCCommands: write memsvc error: Bus error accessing 0x66c0400c
Apr 15 10:45:39 CTP7 local0.info rpcsvc[5827]: amc.readADCCommands: DEBUG: 5: scaADCCommandOutput for OH0 = 0 
```
## Current Behavior

### Steps to Reproduce (for bugs)

1. Checkout code from 834dd83675a3de1239ea6b431d761837fcda7007
2. If you run the module `readADCCommands`. Then one should get
    ```bash=
    ./blaster eagle34 08 0x200
    ```

## Possible Solution (for bugs)

- This comes when one tries to read the disconnected SCA-ADC channel after enabling the current sink. [sca.cpp#L133](https://github.com/cms-gem-daq-project/ctp7_modules/blob/b358ed18fd77ed0625b5f04082df7d19da2d7ff6/src/amc/sca.cpp#L133)
- Once it switch on the current sink then somehow this error persists even for the connected channels until one reestablish connectivity using script `testConnectivity.py`
- The possible solution is to enable the current sink only for the temperature channels not for others. Like
  ```bash=
  // enable/disable the current sink if reading the temperature ADCs
  if (ch == 0x00 || ch == 0x04 || ch == 0x07 || ch == 0x08 || ch == 0x1f)	// Hardcoded the channel numbers
    sendSCACommand(la, SCAChannel::ADC, SCAADCCommand::ADC_W_CURR, 0x4, 0x1<<ch, ohMask);
   ```
- This is already tested and it worked.

## Context (for feature requests)

- This issue was affecting the implementation of #100.

## Your Environment

* Version used: amc v1.0.1
* Shell used: bash

jsturdy, 2019-04-25:
What is a "disconnected" ADC channel (and why are you trying to read them?)

* Can you reset the system and run `testConnectivity.py` and start fresh doing the following
  1) In a loop over "good" ADC channels (after looping over the "good" ADCs, select a "bad" ADC for the last iteration):
      1) read `Imon` in the DCS mainframe
      1) run test on ADC channel
      1) read `Imon` in the DCS mainframe
  1) Then do the following for each ADC channel (first all "good" ADCs, and then one-by-one a "bad" ADC):
      1) reset the system and run `testConnectivity.py`
      1) read `Imon` in the DCS mainframe
      1) run test on ADC channel
      1) read `Imon` in the DCS mainframe
  * log everything in an elog

> The possible solution is to enable the current sink only for the temperature channels not for others. 

1) This is not just a "possible solution", it is the operational mode that *must* be followed.
1) Additionally, the current source *must* **only** be enabled (if needed) for the channel being sampled, and then disabled when done.

ram1123, 2019-04-25:
@jsturdy 
> What is a "disconnected" ADC channel (and why are you trying to read them?)

By "disconnected",  I mean the channels that are not used like [sca_enums.h#L243-L245](https://github.com/cms-gem-daq-project/ctp7_modules/blob/develop/include/amc/sca_enums.h#L243-L245)

jsturdy, 2019-04-25:
> @jsturdy
>
>
> By "disconnected", I mean the channels that are not used like [...]

OK, in this case, it might be good to comment out those enums that are not connected, so that it is *not* possible to enable the monitoring of them.

ram1123, 2019-04-26:
Hi @jsturdy ,

List of good ADCs are: 00, 04, 07, 08, 0E, 0F, 11, 12, 13, 15, 18, 1B, 1E and 1F.
List of disconnected/bad ADC are: 01, 02, 03, 05, 06, 09, 0A, 0B, 0C, 0D, 10, 14, 16, 17, 19, 1A, 1C, and 1D

> * Can you reset the system and run `testConnectivity.py` and start fresh doing the following
>   
>   1. In a loop over "good" ADC channels (after looping over the "good" ADCs, select a "bad" ADC for the last iteration):
>      
>      1. read `Imon` in the DCS mainframe
>      2. run test on ADC channel
>      3. read `Imon` in the DCS mainframe

Performed this test and the value of Imon remains same (i.e. 1.85 A) before and after the test.

>   2. Then do the following for each ADC channel (first all "good" ADCs, and then one-by-one a "bad" ADC):
>      
>      1. reset the system and run `testConnectivity.py`
>      2. read `Imon` in the DCS mainframe
>      3. run test on ADC channel
>      4. read `Imon` in the DCS mainframe
>   

Also, performed this test and the value of Imon remains same (i.e. 1.85 A) before and after the test.

>   
>   * log everything in an elog

Its logged in [cms-elog-1082877](http://cmsonline.cern.ch/cms-elog/1082877)

> 
> > The possible solution is to enable the current sink only for the temperature channels not for others.
> 
> 1. This is not just a "possible solution", it is the operational mode that _must_ be followed.
> 2. Additionally, the current source _must_ **only** be enabled (if needed) for the channel being sampled, and then disabled when done.

closed 2019-04-26 by ram1123

ram1123, 2019-07-26:
This issue was understood but forgot to reply.

The issue mentioned above was arised by the current sink. So, whenever we are reading the temperature channel then before reading it we need to enable the current sink and just after reading it we need to disable it. If we don't disable and try to read any other register then we will get the error saying

```
Bus error accessing XXXX
```

**So, every time before reading the temperature channel we need to enable the current sink and disable it after reading.**