cms-gem-daq-project/ctp7_modules #129: [feature request] Create a remote callable method for confCalPulseLocal

bdorney, 2019-06-11:
## Brief summary of issue

After discussing with @jsturdy we will need to have a remote callable method for `confCalPulseLocal`.  This should go only in the legacy branch, e.g. `release/v1.1.X`.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

https://github.com/cms-gem-daq-project/ctp7_modules/blob/develop/src/calibration_routines.cpp#L39

Have a remote callable method for the local function `confCalPulseLocal`.

## Current Behavior

No remote callable method exists.

## Context (for feature requests)

When trying to develop a script similar to [set_scanParams_latency.py](https://github.com/cms-gem-daq-project/cmsgemos/blob/generic-amc-RPC-v3-short-term/gempython/tools/xdaq/set_scanParams_latency.py) but for scurves via tracking data path we will need a way to start/stop calibration pulses to a given set of channels.

ram1123, 2019-06-14:
Hi @bdorney,

I added the remote callable function as below:

```cpp
void confCalPulse(const RPCMsg *request, RPCMsg *response)
{
    GETLOCALARGS(response);

    uint32_t ohN = request->get_word("ohN");
    uint32_t mask = request->get_word("mask");
    uint32_t ch = request->get_word("ch");
    bool toggleOn = request->get_word("toggleOn");
    bool currentPulse = request->get_word("currentPulse");
    uint32_t calScaleFactor = request->get_word("calScaleFactor");

    confCalPulseLocal(&la, ohN, mask, ch, toggleOn, currentPulse, calScaleFactor);
    LOGGER->log_message(LogManager::DEBUG, stdsprintf("Finished configuring the calibration pulse."));

    rtxn.abort();
}
```
Also, I modified the corresponding header file. 

Its compiling without any error. How I can run this and see if the calibration pulse is configured correctly or not?

bdorney, 2019-06-14:
Reserve time on the coffin setup and use the same approach you used when testing the SCA ADC interface.

You can read back registers using the `gem_reg.py` tool.  e.g. try to use your function to write registers and then use `gem_reg.py` to see if your changes were applied.

commit 3e7670d7 (2019-06-21): Merge pull request #135 from ram1123/feature/v1.1.X_RC_ConfCalPulse

Added remote callable function for confCalPulseLocal

Closes #129
closed 2019-06-21 by jsturdy