cms-gem-daq-project/ctp7_modules #139: Replace the "`LOGGER`" logging system with a full-featured logging library, e.g. `log4cplus`

lpetre-ulb, 2019-07-09:
## Brief summary of issue

The "`LOGGER`" logging system is limited and tightly coupled with the `rpcsvc` daemon. In order to implement some functionalities (standalone programs, log level change at runtime, ...) and to guarantee maintainability it may be time to move to a full-featured logging system.

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Expected Behavior

The following points should be taken into consideration while choosing a new solution:

* The functions and functors provided inside the `ctp7_modules` should be usable in standalone programs.
* The log level should be changeable at runtime. See this issue #134.
* The ATCA `rpcsvc` equivalent might not provide the "`LOGGER`" logging system.

## Current Behavior

The logging system is tightly coupled with the `rpcsvc` daemon.

## Context (for feature requests)

@jsturdy was trying to write standalone programs for the CTP7.

Moreover the templated RPC refactoring starts to decouple the `ctp7_modules` from the `rpcsvc` server so it is the perfect time to change the logging system.

## Proposal

Since `xDAQ` and `xHAL` already use the `log4cplus` library I would suggest to use it as well on the CTP7. It is already compiled for `arm` and it would provide a common logging pattern.

A minimal example which logs to the CTP7 `syslog` daemon is trivial to write :

``` c++
#include <log4cplus/logger.h>
#include <log4cplus/configurator.h>
#include <log4cplus/loggingmacros.h>

int main(int, const char**)
{
    log4cplus::initialize();
    log4cplus::PropertyConfigurator::doConfigure("log.properties");
    log4cplus::Logger logger = log4cplus::Logger::getInstance(LOG4CPLUS_TEXT("main"));

    LOG4CPLUS_WARN(logger, LOG4CPLUS_TEXT("Hello, World!"));

    return 0;
}
```

with the following `log.properties` file:

``` txt
log4cplus.appender.syslog=log4cplus::SysLogAppender
log4cplus.appender.syslog.ident=syslogut
log4cplus.appender.syslog.facility=user
log4cplus.appender.syslog.layout=log4cplus::TTCCLayo

log4cplus.rootLogger=DEBUG,syslog
```

Special care will have to be taken when initializing the library inside the RPC modules since they are all loaded independently. The best place would be inside the `rpcsvc` daemon where the "`LOGGER`" logging system is initialized.

jsturdy, 2019-08-21:
(copied from @lpetre-ulb's comment on mattermost)
Hi @jsturdy, I was starting with the `log4cplus` [release chosen by `xDAQ`](https://gitlab.cern.ch/cmsos/core/tree/master/extern/log4cplus) and its upstream spec file (with additional hints from the `gembuild` template spec file) on the ULB setup. Building native RPMs is trivial and cross-compiling is not more complicated. However when it comes to cleanly cross-building RPMs, it is harder. The are two main reasons:

1. The spec files don't seem to be designed for cross-building (at least not with an external toolchain like the Xilinx's): `BuildArch` cannot be used, the `%_arch` variable/macro is not defined, ...
2. The installation path is different for the CTP7 package and the devel package installed on the development machine. So far, if I don't want to copy files by hand, the only solution I found is to create two spec files and build the library twice.

Nevertheless it more or less works now.

Though, the integration into the `xHAL` build system was not very clear... but I just saw you recently added a lot of documentation. :-) In the end I think we could follow the `xDAQ` structure with its `external` directory. The main question I have left before following that path is the following. Should I mimic the current copy-based system or, what I'd prefer, fully use the spec file with the library compilation done in `%build`?

jsturdy, 2019-08-21:
> and its upstream spec file (with additional hints from the gembuild template spec file)

link please?

jsturdy, 2019-08-21:
a few questions/comments to scope this issue:
1) we're only discussing this for packaging `extern` for the `ctp7`, right (effectively, all the necessary tools that are in the `xcompile` folder)? -- ignore the devel package
2) with the patch provided by @lmoureau, we can now build `xhal` with either 1.2 or 2.0 `log4cplus`, right?
2.a) if that's the case, are features needed from the 2.0 API to get the transition to `log4cplus` in `xhal(arm)`/`ctp7_modules`?
2.b) if not, then just ignore for now the packaging question because there *is* a *compatible* `log4cplus` on both sides, and there is *no* requirement that they be the same version (of course, if this statement is false, we revisit, but then we just address point 1)

lpetre-ulb, 2019-08-21:
> > and its upstream spec file (with additional hints from the gembuild template spec file)
> 
> link please?

I was referring to [this spec file](https://github.com/log4cplus/log4cplus/blob/REL_2_0_4/log4cplus.spec) in the upstream project and [this one](https://github.com/cms-gem-daq-project/gembuild/blob/feature/common-ci/specTemplate.spec) in the `gembuild` repository.

> a few questions/comments to scope this issue:
> 
>     1. we're only discussing this for packaging `extern` for the `ctp7`, right (effectively, all the necessary tools that are in the `xcompile` folder)? -- ignore the devel package

Currently `log4cplus` is compiled in `xhal/xcompile`, manually installed in the Peta Linux sysroot and manually copied to the CTP7. I think we can (should?) do better and provide RPMs for both the CTP7 and the development machine. This is not high priority, though

>     2. with the patch provided by @lmoureau, we can now build `xhal` with either 1.2 or 2.0 `log4cplus`, right?

As soon as it is merged, yes.

>        2.a) if that's the case, are features needed from the 2.0 API to get the transition to `log4cplus` in `xhal(arm)`/`ctp7_modules`?

Since I don't want to create loggers by hand, but prefer to use a configuration file, the 1.2 and 2.0 APIs look compatible.

>        2.b) if not, then just ignore for now the packaging question because there _is_ a _compatible_ `log4cplus` on both sides, and there is _no_ requirement that they be the same version (of course, if this statement is false, we revisit, but then we just address point 1)

Indeed there is no requirements that they share the same version. However I think it is better to start with a library that fully supports C++11 and above.

lpetre-ulb, 2019-08-21:
After a few benchmarks, `log4cplus` appears to be as fast as the `LOGGER` system. Even slightly faster and that could maybe be improved with a multi-threaded `log4cplus` build. Messages with a log level less than current log level have a near to zero cost, so providing a lot of debug information is possible.

Also, the library is loaded only once for all RPC modules. Unfortunately the logger hierarchy must be reinitialized for each module.

Before creating a PR, I would have a few questions :

1. Where should the configuration file be located? I was thinking of `/mnt/pesistent/gemdaq/etc`.
2. How do we define the logger names? Since `log4cplus` works with a hierarchy of loggers we could leverage that feature. For example, it allows different log levels for different parts of the program.
3. Finally, before porting all the modules to `log4cplus`, I think we should define a [policy for the log levels](https://github.com/cms-gem-daq-project/ctp7_modules/issues/134).

jsturdy, 2019-08-22:
> After a few benchmarks, `log4cplus` appears to be as fast as the `LOGGER` system. Even slightly faster and that could maybe be improved with a multi-threaded `log4cplus` build. Messages with a log [...]

What are the implications of this?  Is this fundamentally different than the current `LOGGER` system?

> 
> Before creating a PR, I would have a few questions :
> 
>     1. Where should the configuration file be located? I was thinking of `/mnt/persistent/gemdaq/etc`.

This is reasonable to me.  Do we envision the log config being packaged with the `xhal` library, the `ctp7_modules`, both?  Is the expectation that this log config will be modified on the card outside of a package update?  Is there a hierarchy of locations that `log4cplus` will look for a config?

> 
>     2. How do we define the logger names? Since `log4cplus` works with a hierarchy of loggers we could leverage that feature. For example, it allows different log levels for different parts of the program.

Can you post some output of logs coming in from various places?
What were you thinking viz the names?
If I understand the model correctly, each time a logger is created, it would have the option of creating a new logger (with a potentially different log level), or obtaining an existing logger from the service (using the set log level).

> 
>     3. Finally, before porting all the modules to `log4cplus`, I think we should define a [policy for the log levels](https://github.com/cms-gem-daq-project/ctp7_modules/issues/134).

Is [this](https://github.com/cms-gem-daq-project/rpcsvc/blob/f75364532253c93dbadece18b8be0fac7140236d/LogManager.h#L16-L25) not sufficient?
In [`cmsgemos`](https://github.com/cms-gem-daq-project/cmsgemos/blob/develop/gemutils/include/gem/utils/GEMLogging.h#L10-L15) there are fewer levels, but the policy is effectively the same.

jsturdy, 2019-08-23:
> Still about `xhal`, we might want to change the logger initialization, at least for the `XMLParser`, since currently the log output to the console is hard-coded.

Agreed.

> Yes, it can happen during a debugging session for example. The default configuration might be to send logs to the local `syslog` with a log level of `INFO`, but during debugging one might want to store the logs up to `DEBUG` into a local file.

Sure, but this can be also controlled at run-time (well, maybe less easily over RPC...) a la how the `xdaq` process changes log level on-the-fly.  But this question was really only for packaging purposes, so isn't critical for getting this done.

> Ok, but what is the policy? The meaning of each log level seems clear, but what information must always be logged in `INFO` for example?

OK, better understand your question now.
I don't think that having this policy written in stone is critical (at least for the initial stage); I don't have a strong feeling for mandating what *must* appear in an `INFO` message.  I think it's not unreasonable to hone this over time (I don't expect that developers will wildly abuse the current levels with the info listed in the two links above, and bad behaviour can be corrected in PR review and added to guidelines)

I think I've parsed your reply and responded to your questions (in general, I have no objections to your proposal).  If there's anything further, please draw my attention to it

PR #151 (2019-09-19): Add `log4cplus` initialization to the RPc modules
As proposed in #139, this PR enables the use of the `log4cplus` logging system in the RPC modules.

## Description

At module initialization, the `log4cplus` system is initialized and its configuration reloaded. The configuration is read from `/mnt/persistent/gemdaq/etc/rpcsvc_log.properties` with a fallback to an embedded  default configuration. Therefore the different `log4cplus` logging macros can now be used through the code.

The default embedded configuration sends the logs from the `INFO` log level to the local `syslog` daemon with the p [... cut]
commit 1d8a7d01 (2019-09-19): Merge pull request #151 from lpetre-ulb/feature/templated-rpc-methods

Add `log4cplus` initialization to the RPC modules

Closes #139
closed 2019-12-06 by mexanick