cms-gem-daq-project/ctp7_modules #145: amc.h does not include definition of an amc class called in src/calibration_routines.cpp

vetens, 2019-07-15:
## Brief summary of issue
When src/calibration_routines.cpp is called, which in turn calls a function amc::OH_PER_AMC, I receive an error stating that 'amc' has not been declared.

### Types of issue

- [x] Bug report (report an issue with the code)
- [ ] Feature request (request for change which adds functionality)

## Expected Behavior

## Current Behavior

error raised when attempting make rpm

### Steps to Reproduce (for bugs)

1. run ``make rpm``
2.
3.
4.

## Possible Solution (for bugs)

There is an amc.h in include that does not have the definition of an amc class in it so I suspect there is a newer version somewhere with that definition that just hasn't been merged yet
## Context (for feature requests)

## Your Environment

* Version used:
* Shell used:

jsturdy, 2019-07-15:
Can you add the git hash you're compiling against to the `Your Environment` section?

vetens, 2019-07-15:
updated: ``develop 2c5d65f``

jsturdy, 2019-07-15:
Thanks!
Indeed, unfortunately the backport hotfix (#125) didn't take into account that these had moved around...
If you want to do a patch of this you will need to simply add:
```c++
#include "hw_constants.h"
```
to the includes in `calibration_routines.cpp`

vetens, 2019-07-15:
Thanks! That worked.

closed 2019-07-17 by vetens