cms-gem-daq-project/ctp7_modules #158: Emit RPC error specifics

vetens, 2019-11-11:
## Brief summary of issue
This issue is for the discussion of what to do with specific ``EMIT_RPC_ERROR`` functions which are called.

Since we are migrating the module to not make explicit reference to the rpc methods we will need to rethink any times where the code explicitly emits an rpc error and perhaps raise an exception

### Types of issue

- [ ] Bug report (report an issue with the code)
- [x] Feature request (request for change which adds functionality)

## Current Behavior

 
For example, inn ``gbt.cpp`` lines 46, 107, 125, 129, 162, 166, 189, and 193, there are several calls to ``EMIT_RPC_ERROR( response, message, errorcode)``. For some of these we may want to raise an exception in addition to logging it in the usual manner.
Currently the implementation for this migration is to replace the ``EMIT_RPC_ERROR`` calls with ``LOG4CPLUS_ERROR(logger, message)`` so I would like to establish how and when to raise exceptions in ``gbt`` and other submodules

jsturdy, 2019-11-11:
In the cases that these errors `return` exit codes that indicate failure, now and appropriate exception should be raised (`std::runtime_error`, `std::range_error`, etc.,, though @lpetre-ulb or @lmoureaux can comment if there only certain types that will pass natively through the template)
see [here](https://en.cppreference.com/w/cpp/header/stdexcept)

vetens, 2019-11-11:
so in one of those cases should I still also log it to ``LOG4CPLUS_ERROR``?

jsturdy, 2019-11-11:
I think all exceptions should probably be logged as error or warning at least, as a general rule